PLCHome/node-red-contrib-ads #3: Connection problem

scelle, 2018-08-22:
Hi, 

still no succes using the ads node.  The socket connection seems to be working, but then I get several timeout messages and a reconnection attempt is made.
My plc is a TC2 runtime on a windows CE system. 
following the documentation, the "create symbols" option should be enabled on 2 places:
- plc task: not possible in my case:
[image]
(Alternatively, in the I/O Idle task this is possible) 
- Device image: enable ads Server + enable 'create symbols'
[image]

log messages:
[image]

Do I need to install something else on the target system (in my case windows CE)? (Twincat ADS webservice?)

Any more suggestions?

kind regards 
Pascal

ChrisHanuta, 2018-08-22:
[image]

Did you insert the static route?
The port is in the picture on 802.

ChrisHanuta, 2018-08-23:
If I understood correctly, it is on the router on the CE PLC. I found this:

1. TwinCAT AMS Router doesn't allow multiple TCP connections from the same host. So when you use two AdsLib instances on the same host to connect to the same TwinCAT router, you will see that TwinCAT will close the first TCP connection and only respond to the newest.
2. As ADS is transmitted over a TCP connection, there is no real time guarantee.
 
You can set up a second IPv4 on the PC and assign to this a ADS NET ID under Twincat. But you have to disable DHCP and enter all IP addresses by hand.

ChrisHanuta, 2018-08-27:
readme revised