PLCHome/node-red-contrib-ads #4: Need help to connect on local machine.

yannickasselin, 2018-09-25:
Nice work,

I tested it with TwinCAT 3 and it works fine when node red runs on a seperate machine than the TwinCAT runtime.

I am now trying to run node red on the same machine as the TwinCAT runtime but cannot figure out how to setup the "Host", "Target NetId" and "Source NetId".
I tried to put 127.0.0.1 for the host and the same AmsNetId for both Target and Source but it does not work.

Thank you

ChrisHanuta, 2018-09-25:
This code only uses the ADS specification. The Beckhoff dll communicates with the local router.

Unfortunately, I only have TC2 PLC. Normally it should have a second netid which has no 0 at the end. I had recently debated the topic.

[I hope this answers the questions:](https://github.com/roccomuso/node-ads/issues/8)

yannickasselin, 2018-09-26:
I tried a lot. Still does not work on the same machine. :(
I must be missing something...

yannickasselin, 2018-09-26:
I tried adding a second route on the same IP Address with a different NetId (Just like acidos does in the other thread) and it still does not work. I noticed that when adding the second route, it does not show "Sub Route" in the comment section. Could it be the problem?

yannickasselin, 2018-09-26:
Here are the steps:

[image]

ChrisHanuta, 2018-09-27:
Under TC2, the first route actually already exists, since this is needed for the configuration, I am surprised that this must be entered.
Ever tried to set up a second IP address, dhcp must be turned off.

I have no time this weekend. But I can try next week to install TC3.

yannickasselin, 2018-09-27:
The first route was already there but I deleted it to start from scratch and show you all the steps.
I will try to find some time and test it with TC2 just to see if I get the same results.
By the way, you can install TC3 on the same machine as TC2 and you can even switch between runtimes. I have not done that in a while but I remember doing it years ago. I have worked a lot with TC2 in the past but have been working exclusively with TC3 for the last 3 years.

yannickasselin, 2018-09-27:
I tested it with TC2 installed in a virtual machine and it works fine. Even with DHCP enabled. I noticed that when I create the second route, "Sub route" is written in the comment column. I don't get that when creating a second route in TC3.
I then installed TC3 in the same virtual machine and tested it again with node-red, still does not work. :(

ChrisHanuta, 2018-10-10:
There seems to be a problem with the routing. I see the requests to the local router, the answers are not converted to ADS-IP. ADS sends it back to the PLC on the internal connection. If I set up as a workaround two static IP addresses it works for me. Unfortunately I can only use the ADS specification. I also do not know why this works in the last version with TC2.

closed 2018-10-10 by ChrisHanuta

yannickasselin, 2018-10-10:
Thanks, I was finally able to make it work by adding a second IP Address. But I had to use the second IP Address in the HOST field.

[image]

It works! And by adding a second IP address, I am also able to use the system manager at the same time as node red runs. Cool stuff!

Thanks a lot.

ChrisHanuta, 2018-10-10:
[I would be happy about a vote](https://flows.nodered.org/node/node-red-contrib-ads)