PLCHome/node-red-contrib-ads #5: Multiple notifications

o5i, 2018-09-30:
Hi, nice project..
Im trying to replace a damaged touch panel by setting up a node red web visu.
Anyway, i try to read multiple values by multiple notifications but some just doesent comunicate. I saw they did something on the api to prevent read timeouts. Im not sure how i should read the values now. I have tried to trigger the ADS IN manualy but that doesent work. On the api they have the Multiread function, i think i need something like that, or im not sure if that works. Anyway, is there a workaround to read the values? Cheerz

ChrisHanuta, 2018-09-30:
Multiread is a special ADS communication that is only supported in newer PLCs. In principle, it makes no difference whether a multiread or single read is performed. only the network load increases.

- First, the ADS route to due PC with NODE-Red must be entered in the PLC.
- Node-Red is not supported on Windows. 
- The ADS communication on the local PLC does not work especially well.

ChrisHanuta, 2018-10-01:
Let's start from scratch.

- Twincat version.
- Node Red Version
- Operating System for Node Red

I currently have no problem with the notification. Currently 140 to Twincat 2.11 under Docker with Jessie on RPI3.
Have you ever tried to read the variable list with the node?

o5i, 2018-10-01:
twincat version is 2.11
node-red 0.19.4
nodejs 8.12.0

but i have tried before on a debian vm with nodejs 10 same result.
what you mean with read variable list with the node?

ChrisHanuta, 2018-10-01:
Do you use >1.1.12:

> 1.1.12: Maintenance Release
> 
> Enhancements
> 
>     Example for web ui with Node for get symbols and types from ADS.
> 
> Fixes
> 
>     ADS Notification Note wont work
>

o5i, 2018-10-02:
Im using 1.1.12 from the node-red palette/install menu.. i have to check
later for the list.. cheerz

o5i, 2018-10-02:
Strange thing is that there is no error at all. It just doesent return
message. How to debug?

o5i, 2018-10-02:
> Exactly, but i need to try bit more.
> The debug call is:
> DEBUG=node-red-contrib-ads,node-ads node node_modules/node-red/red.js ....
> The current debug issues do not help. I'll add a few more in the next few
> days.
> But you see no exceptions on the bash?
> And you get a few values ​​but not all?

hi, i have tryed the example with the symbol list but it doesent work...
node red just crashes...
t 20:55:14 - [red] Uncaught Exception:
2 Oct 20:55:14 - ReferenceError: n is not defined
    at EventEmitter.<anonymous>
(/root/.node-red/node_modules/node-red-contrib-ads/ads-connection.js:321:17)
    at Object.cb
(/root/.node-red/node_modules/node-ads-api/lib/ads.js:629:12)
    at Object.<anonymous>
(/root/.node-red/node_modules/node-ads-api/lib/ads.js:960:15)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)

ChrisHanuta, 2018-10-05:
I think it's the connection. Did you put the connection on the node? If so, please set the ADS system node and a debug node on the flow.
(Menue/Import/Clipboard and choose your connection)
I think you have an problem with the ADS route and the ADSnetID in the PLC.

`[{"id":"22de625.16e9e9e","type":"ADS System","z":"c26e486b.2f1238","name":"","datasource":"","x":550,"y":240,"wires":[["910fe95f.b47c18"]]},{"id":"910fe95f.b47c18","type":"debug","z":"c26e486b.2f1238","name":"","active":true,"tosidebar":true,"console":true,"tostatus":true,"complete":"payload","x":980,"y":240,"wires":[]},{"id":"b6f0c9aa.c1db58","type":"inject","z":"c26e486b.2f1238","name":"","topic":"1","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":310,"y":240,"wires":[["22de625.16e9e9e"]]}]`

ChrisHanuta, 2018-10-10:
Please check the Version 1.1.13.

ChrisHanuta, 2018-10-30:
Multiread is not necessary. There must be another problem here. But I am currently working on a solution for types and multiread.

closed 2018-10-30 by ChrisHanuta

o5i, 2018-10-30:
Ok.. i havent tryed yet as the project changed.. i will see to check that
soon as posdible..