PLCHome/node-red-contrib-ads #6: Node.js under heavy load when connection down

scelle, 2018-10-08:
configuration: 
OS: windows server 2016
Node-Red: v 0.19.2
PLC: Twincat 3

Hi, 
PLC is regulary offline (=normal situation) . In that case Node.js seems te be very busy trying to re-establish the ads connection.  So busy that I even can't use Node-red anymore (UI gets no more response from node-red)
2 questions:
- is this a normal situation?
- if yes, how can I disable/slow down the reconnection attempts?

kind regards
Pascal

ChrisHanuta, 2018-10-08:
I'm surprised about this behavior.
The node should actually try to restart after 45 seconds if the ADS component does not respond.
If an error occurs, the NODE should wait 2 seconds or 20 seconds.

Can you post a part of the log or the cmd output?
Maybe I can see what's happening there.

If the PLC is running you have connection?

ChrisHanuta, 2018-10-08:
It is up to the windows implementation of the node.js net.
Under windows connect is called although the socket triggers an error.
As a result, the connect performs too many actions.
Let's see how I get the handle

```
8 Oct 22:37:47 - [info] Starting flows
  node-red-contrib-ads connect: start connect +0ms
  node-red-contrib-ads startTimer: 45000 +0ms
8 Oct 22:37:47 - [info] Started flows
  node-red-contrib-ads connect: callback  undefined +132ms
  node-red-contrib-ads startTimer: 2000 +3ms
  node-red-contrib-ads error: node.adsClient.on  Error: target machine not found
    at getError (d:\rasppi\node-red\node_modules\node-red-contrib-ads\node_modules\node-ads-api\lib\ads.js:1666:13)
    at Object.emitAdsError (d:\rasppi\node-red\node_modules\node-red-contrib-ads\node_modules\node-ads-api\lib\ads.js:1657:13)
    at Object.analyseResponse (d:\rasppi\node-red\node_modules\node-red-contrib-ads\node_modules\node-ads-api\lib\ads.js:205:16)
    at Object.checkResponseStream (d:\rasppi\node-red\node_modules\node-red-contrib-ads\node_modules\node-ads-api\lib\ads.js:190:25)
    at Socket.<anonymous> (d:\rasppi\node-red\node_modules\node-red-contrib-ads\node_modules\node-ads-api\lib\ads.js:124:25)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
    at readableAddChunk (_stream_readable.js:250:11)
    at Socket.Readable.push (_stream_readable.js:208:10) +16ms
  node-red-contrib-ads error: node.adsClient.on  1 +4ms
8 Oct 22:37:47 - [error] [ads-connection:3fd68a9e.3cb1b6] Error ADS: Error: target machine not found
  node-red-contrib-ads error: delete (node.adsClient) +1ms
```

ChrisHanuta, 2018-10-09:
Yesterday I found a solution.
If I manage to test, I can build a version today or tomorrow.
But I'm wrong, it's not a difference between windows and linux. 
It depends on whether the computer is reachable and no ADS is available or the socket can not connect.

ChrisHanuta, 2018-10-09:
1.1.13 published

closed 2018-10-09 by ChrisHanuta

ChrisHanuta, 2018-10-10:
[I would be happy about a vote](https://flows.nodered.org/node/node-red-contrib-ads)