PLCHome/node-red-contrib-ads #7: Use of Topic in ADS notification node

scelle, 2018-10-17:
feature request
It would be nice if there is an option to define the topic for an ads notification node.  
In the current situation you receive a message with just a value, so with no relation to which variable this value belongs to.
If input field Topic is left blank the name of the variable could be used by default.

ChrisHanuta, 2018-10-17:
Yes i can do that. 
At the moment you can set only an property name: msg.payload.PropertyName = value

I have a look in the next two weeks

ChrisHanuta, 2018-10-17:
Topic added on all nodes

ChrisHanuta, 2018-10-19:
is that what you need? 
Or did I misunderstand the request?

scelle, 2018-10-20:
Perfect like that!. Ideal for logging etc... Thanks for the fast response! This ads-node is a very helpfull tool for easy accessing twincat plc's!

ChrisHanuta, 2018-11-11:
It works for me. 
Which node did you try?
Did you enter the topic, otherwise the tropic will not be send.

martinschki, 2018-11-11:
oh, i thought its sending the variable name if topic is kept empty!

ChrisHanuta, 2018-11-11:
oh sorry, it does not work that way.

closed 2018-11-11 by ChrisHanuta

martinschki, 2018-11-11:
Sorry, I assumed since it was in the original ticket at the top
no worries! great node!!

scelle, 2020-04-30:
hey, 

I noticed that using the topic in the symbols node doesn't seem to work similar like the ads in node.
on the ads-in node, if you set msg.config.topic this value will be transferred to  msg.topic at the output of the node.
For symbols that doesn't seem to work.