PLCHome/node-red-contrib-ads #8: No Connection to CX9010

martinschki, 2018-11-05:
Hello,

I've added the static route and I had at one point in time ads with my PLC already running (via the TAME 4 lib https://github.com/tomcx/tame4)
Now I get:
"Error ADS: Error: connect ECONNREFUSED 192.168.1.10:48898"

I double checked IP addresses and ports...

Any ideas?

ChrisHanuta, 2018-11-07:
For the analysis I need some information:
Twincat version?
Same / Other PC as Twincat PLC running?
More then one Twincat Cliends on one IP address?
TWINCAT system monitor running?

The ADS Net ID in the PLC setting must match that in Node Red (Source NetId). -> "Error on connect: check target NetId or routing"
The PLC ADS Net ID must also be correct (Target NetId). -> "Error on connect: check target NetId or routing"
The IP of the PLC must be exist correct -> "Error ADS: Error: connect ETIMEDOUT XX.XX.XX.XX:48898"

The error message comes from the TCP Soket and not from the ADS implementation on 192.168.1.10 port 48898, no connection can be established. I get this message when I my router fed the PLC "Error ADS: Error: connect ECONNREFUSED XX.XX.XX.XX: 48898"
It looks as if Beckhoff Twincat ADS on the 192.168.1.10 is not running

martinschki, 2018-11-10:
Twincat: 2.11.2228
other, it is a CX9010
and ads running on RaspberryPi
only one twincat client
system monitor not running (and if, on other IP address)

martinschki, 2018-11-11:
it is running now....
In restartet node red and also Twincat a couple of times...
I'm (pretty) sure I didn't do anything different this time...

closed 2018-11-11 by martinschki