PLCHome/node-red-contrib-ads #9: Error ADS: Error: write after end

scelle, 2018-11-13:
node-red v0.19.2
node-red-contrib-ads version 1.1.14
Twincat 3

Hi, 
I have a simple config, using 3x ADS_IN node for 3 different variables, calling the node cyclic every 0.5s
This worked fine for a while, but now I'm getting the following error at a rate of almost 2 times per second.:
Error ADS: Error: write after end

I checked the Node-red logging:

12 Nov 18:10:46 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:46 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:46 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:47 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:47 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:47 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:47 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:51 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:51 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:51 - [error] [ads-connection:7d3494b1.bd65dc] Ads read timeout
12 Nov 18:10:57 - [error] [ads-connection:7d3494b1.bd65dc] Error ADS: Error: read ECONNRESET
12 Nov 18:10:58 - [error] [ads-connection:7d3494b1.bd65dc] Error ADS: Error: write after end
12 Nov 18:10:58 - [error] [ads-connection:7d3494b1.bd65dc] Error ADS: Error: write after end
12 Nov 18:10:59 - [error] [ads-connection:7d3494b1.bd65dc] Error ADS: Error: write after end
12 Nov 18:10:59 - [error] [ads-connection:7d3494b1.bd65dc] Error ADS: Error: write after end

At the moment I have no idea why I got ADS read timeouts.  But the "write after end" error occured after doing the connection reset. 

Any idea?  I found this https://stackoverflow.com/questions/32274103/node-js-sharp-library-write-after-end-error
But I can't make the connection to my problem.

ChrisHanuta, 2018-11-13:
There are still some problems with the ADS-API implementation. It's not mine. I just expanded it. The thing would have to be redesigned.
A TCP socket object is created for the connection. The error says that was written to the object after the TCP socket was closed. The TCP Soket will be closed automatically when an "Error ADS: Error:" message comes up.
Please try the Notification object for cyclic reading. That is desigt for it: cyclically or on event.
I can not say why the PLC generates a timeout. There seems to be a problem with TC3 or it's because of the local installation of node-red.
That works on the local machine, right?

scelle, 2018-11-13:
Hi, 
No local installation.  connection between PLC and NR is over LAN.  There's a realistic chance that the primary cause was a network interruption.  
Before I tested the Notification object already.   The reason why I switched to ADS_IN was that the Notification node didn't seem te react as I expected.  Changing cycle time didn't seem to result in a changing behaviour.  
Anyhow, I tested it again.  At first I got some Error ADS .  After restarting Node-Red, the Notification node works fine.  I'll keep using this config and do some follow-up.

ChrisHanuta, 2018-11-14:
"Ads read timeout" is generated when a request is sent to the PLC and the PLC does not respond within 500ms. The time can be adjusted in the node-ads-api/lib/ads.js on line 990.

"ECONNRESET" means the other side of the TCP conversation abruptly closed its end of the connection. This is most probably due to one or more application protocol errors. Maybe you can see something in the logs from the SPS.

Error ADS: Error: write after end: Is a following error, because the connection is no longer established. normally the connection is disconnected and rebuilt. To find out why this is trying to write in the connection I would have to see the logs

`set DEBUG=node-red-contrib-ads:*,node-ads*`
before starting NODE-RED

one way to find out if there are network problems is the software Wireshark. Install, click on the black button in the bar. Select network card. Enter the filter in the green line: ip.addr eq <PLC IP>
If you see black lines then there are transmission errors.

ChrisHanuta, 2018-11-15:
With the timeout is also a known problem when the system monitor for twincat 2 runs on the same system as nodered. This is a Twincat router bug in the newer versions. Do you have the twincat 3 development interface running? Then the error would be synonymous in twincat 3. You can see it in wireshark. The data packets sendet to the wrong port.

closed 2018-11-24 by PLCHome