PLCHome/node-red-contrib-ads #10: Web example for "types" crashes node red

martinschki, 2018-11-14:
When I type an "*" inot the search field of the Types field, node red restarts.

Same thing at the variable field works just fine

ChrisHanuta, 2018-11-15:
weird, i just tested it with 1.1.14 and works fine.
Can you look into the cmd log. Possibly stop the demon and start on the cmd.
And enable the DEBUG: on WIN: `set DEBUG=node-red-contrib-ads:*,node-ads*` before starting NODE-RED, on Linux `DEBUG=node-red-contrib-ads:*,node-ads* node ....`

And information about the operating system and node version

ChrisHanuta, 2018-11-15:
please look at the datasources. Not that there is a double

martinschki, 2018-11-15:
enabled debugging restarted from command line, not I can't reproduce it anymore...
(scratching my head...)
please delete if you feel appropriate

closed 2018-11-15 by martinschki

ChrisHanuta, 2018-11-15:
no problem, should it happen again: feel free to reopen

martinschki, 2018-11-15:
thank you!
I can't tell you how much I appreciate this node!!
It finally enables me to connect my CX9010 to nodered and get properly started with my home automation
If you need someone to test stuff please let me know