PLCHome/node-red-contrib-ads #11: Notification question

martinschki, 2018-11-14:
Sorry for posting a question here:
You state that Beckhoff says there should be no more than about 500 notifications registered on a plc.

If I create multiple notifications with the same variable on the same or different flows, will those count as one notification on the Beckhoff side?
In other words, if I need the status of one variable on multiple flows how many times does that count for towards that 500 notification limit?

ChrisHanuta, 2018-11-15:
No that does not count in the 500 notification limit, since no registration takes place on the SPS. If you register Nodes with different attitudes, always the first Node registered on the SPS. But it would be better to use a link in and link out node.

ChrisHanuta, 2018-11-15:
measures, I've just seen the code. i think it does not work with multiple notification. Each node is registered individually in the PLC, I had changed that once.
I have to look, that can be changed.

closed 2018-11-18 by ChrisHanuta