PLCHome/node-red-contrib-ads #12: address bit in variable

martinschki, 2018-11-24:
Is there a way to address a single bit from a variable?
e.g. MAIN.variable.0 where MAIN.variable is a variable with type "WORD" in the PLC
It works in the PLC, it would be awesome also here...

PLCHome, 2018-11-24:
Maybe in TW3 if you overload an type pointer like:
TYPE ST_ControllerData :
STRUCT
    nStatus_OperationEnabled : BIT;
    nStatus_SwitchOnActive   : BIT;
    nStatus_EnableOperation  : BIT;
    nStatus_Error            : BIT;
    nStatus_VoltageEnabled   : BIT;
    nStatus_QuickStop        : BIT;
    nStatus_SwitchOnLocked   : BIT;
    nStatus_Warning          : BIT;
END_STRUCT
END_TYPE 

There nothing in the ADS documentation about this.
Would be glad if you could ask Beckhoff support the question.
support@beckhoff.com

martinschki, 2018-11-24:
will do & no TC3 here...

closed 2019-01-31 by PLCHome