PLCHome/node-red-contrib-ads #14: ADS notification not working after PLC has been offline

scelle, 2019-02-04:
Hi, 

I'm using v1.16 of node-red-contrib-ads and twincat 3
I'm having problems after the PLC has been offline.  
When restarting the plc, ads connection is established again:
- I get connectstate 2 (Connected) and adsstate 5 (running).  
- I can write variables with ads out node

But the ads notification node doesn't start sending data.
Only when doing "Full deploy" or restarting node-red solves the problem.

[image]

I use ads notification in cyclic and onchange mode
any ideas?

PLCHome, 2019-02-04:
sounds wrong

scelle, 2019-02-08:
Hi, 
I isolated the issue and made a little testproject in a twincat 2 project to check if the problem remains.  unfortunately it does.  For testing purpose I added an ads-in node on the same variable and this one works fine.  So the problem seems to be only the notification node.  
as a temporary solution I can change all my notifciation nodes to ads-in nodes, but I was hoping you would have time to find a  solution for this.

Pascal

scelle, 2019-02-22:
Hi, 

I'm stuck here.  my temporary solution of using ads-in nodes instead of notification nodes is not an option due to this other issue: https://github.com/PLCHome/node-red-contrib-ads/issues/9.
So neither ads-in or ads notifcation is without problems when my plc is switched off and switched on again after a while.
kind regards
Pascal

scelle, 2019-03-04:
As far as I can debug, something seems to be going wrong when unsubscribing the ads-notification.  Unsubscribe function is not triggered in some cases.

Went back to version 1.1.14 and there I do not have the same problem.

closed 2019-03-09 by PLCHome

hbTecAG, 2019-03-26:
Hello

Have the same problem. After a reboot of the CX (5020) the Notification Node does not work anymore ..
What i can do?
Even if the variables change in the PLC, the changes do not get into the Node-RED

Please mark as open Issue

Version: 1.1.20 (Updated today)

Log: 
`6 Mar 11:46:25 - [info] Starting flows
26 Mar 11:46:25 - [info] Started flows
26 Mar 11:47:23 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: read ECONNRESET
26 Mar 11:47:25 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: connect ECONNREFUSED 10.1.121.1:48898
26 Mar 11:49:15 - [info] Stopping flows
26 Mar 11:49:16 - [info] Stopped flows
26 Mar 11:49:16 - [info] Starting flows
26 Mar 11:49:16 - [info] Started flows
26 Mar 11:49:30 - [info] Stopping flows
26 Mar 11:49:31 - [info] Stopped flows
26 Mar 11:49:31 - [info] Starting flows
26 Mar 11:49:32 - [info] Started flows
26 Mar 11:50:06 - [info] Stopping flows
26 Mar 11:50:07 - [info] Stopped flows
26 Mar 11:50:07 - [info] Starting flows
26 Mar 11:50:08 - [info] Started flows
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
[... 44 lines cut ...]
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
[... cut ...]
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found
26 Mar 11:50:27 - [error] [ads-connection:64632e0d.6f1a5] Error on connect: check target NetId or routing
26 Mar 11:50:50 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: connect ECONNREFUSED 10.1.121.1:48898`