PLCHome/node-red-contrib-ads #15: Fetching name of program?

EsbenLorenzen, 2019-02-23:
I am setting up a dashboard to browse available ADS symbols, which is working fine.

When fetching the variable names as objects via the ADS Symbols node, they are return as ".TEST_VARIABLE_1" for example. This is despite the variable being declared inside the MAIN program on the PLC.

**Is there a way to get the relevant program name**, together with the variable name?
If this is not possible at this time, would it be possible to implement, or is this a restriction in the ADS protocol?

PLCHome, 2019-02-24:
".TEST_VARIABLE_1" must be globally defined. 
Programs have no "." at the beginning of the name. 
[image]

Function blocks are types:
[image]
Varname:  CALENDER.FBCONVERTTIME.IN

Types are types:
[image]
Varname:  CALENDER.CALENDAR.UTC

You can also search deeper in the type by looking for the type below the variable. All names must be concatenate with ".".
Varname:  CALENDER.FBCONVERTTIME.TZINFO.BAIS
[image]

closed 2019-02-24 by PLCHome

PLCHome, 2019-02-24:
[image]

I use "? *. *" to find programs

scelle, 2020-03-16:
If I understood the documentation, it should be possible to search multiple symbols in the  ads symbols, in 1 request.  As input I should use a "string array".  How should this be formatted exactly?
I don't find the right syntax...