PLCHome/node-red-contrib-ads #16: Plans to implement MultiRead functionality

EsbenLorenzen, 2019-02-26:
Is there any plans to implement the ability to pass multiple symbols to something like a ADS Multiread node, and receive the corresponding values back?

I see it's alread implemented in the node-ads module, but I don't see it implemented for this node-red adaptation.
I thought I would give it a try myself, however my abilities in javascript are very limited, as I have gotten stuck.

I tried in essence copying the ADS In node in structure (and some code), passing in an array of objects, with the aim to overwrite the node configuration, but getting various errors, which I attribute to not quite being sure how the node-ads module is being used.

**Is this feature in the pipeline at the moment?**
If not, any insights on the best way to do this would be greatly appreciated!

My current workaround is to have a whole bunch of ADS In nodes (as in 50+ nodes), and using a function node to split everything up, and only use as many as needed. This seems very inefficient though, and I would like to make it a bit more dynamic.

PLCHome, 2019-02-27:
Currently I do not plan to implement a multiread.
I do not see how this is to relieve the PLC. If you want to do something with the data you have to poll the data regularly. That does not relieve the PLC in my opinion.

It is possible to read whole structures or to put a notification on them.
I plan to read structures and to notification. The problem with reading structures is the definition in the surface. I do not have much time to worry about it right now.

Basically, structures are already possible at the moment. You just have to read them in RAW and then extract the values ​​with a function node.

EsbenLorenzen, 2019-02-28:
Thanks for you reply.

I understand you point about avoiding to stress the PLC. I'm trying to design something as non-intrusive as possible, so longer scan-times whould be avoided. Although probably not possible to avoid completely.

I was reading through Beckhoffs documentation on the ADS-DLL library, and this document talks about an improved ADS-read function, which can scan 500 variables in a single call. [https://infosys.beckhoff.de/content/1031/tc3_adssamples_c/html/tcadsdll_api_cpp_sample17.htm?id=5354235567466068335](url)
I could not find anything similar mentioned on the javascrip ADS-api, so I am unsure how that is handled there.

Reading a whole structure as RAW is the way to go at the moment then, I have gotten it to rebuild the structure in node-red. Thanks for the suggestions!

closed 2019-03-09 by PLCHome

scelle, 2020-03-04:
Hi, 

Personally, I also think this would be a very nice feature.  Especially if you want to keep your  flow flexible and clean.  I think there's a lot of similarities with the node-red-contrib-iiot-opcua node.  There you have the possibility to inject your opc tags as a complete list.
I assume this will be a big effort to implement.
All understanding if this is not a priority at the moment.
Thanks for the work so far.