PLCHome/node-red-contrib-ads #17: high CPU Load when value of a variable Changes

hbTecAG, 2019-03-05:
Hello

I have the problem that often with a variable change on the PLC, the library causes a CPU usage of over 100%. Usually only a reboot will help.

There are no error messages in the console. Where can I find a log where to help debug?

What can I do there? So it is not usable :-(

It runs on a Raspberry Pi 3B +, TwinCAT 2.11, CX5120.

EsbenLorenzen, 2019-03-05:
By CPU usage of over 100%, is that on the Raspberry Pi, or on the PLC?

What nodes are you using to recieve the value in node-red?
Are you certain it's the node-ads library and not other nodes that are causing the issue?
If you set up a new flow, and start from scratch, does the same issue occur? 

With 100% CPU usage, I would suspect an infinite loop. Or maybe you subcribed to On-Change notification, which changes rapidly?

Extra debug-info can be enabled in the settings.js file:
https://nodered.org/docs/user-guide/logging

hbTecAG, 2019-03-05:
The CPU of the Raspberry Pi goes high. The PLC continues to operate normally.

I am using the ADS Notification node. (Currently 54x and 1x ADS system)

You gave me an idea. I may have found something. But it has to test something else. Probably the node from somewhere recognizes a change of the variable. (Although none is there).
In Debug Node the string is always exactly the same. Nevertheless, the variable is queried and passed on. i.e. that a new value is probably read in every PLC cycle. But I can not imagine why that is ..

Do you have any idea?

PLCHome, 2019-03-05:
> ADS Notification node. Currently 54x 
I believe that there is a notification "crazy". 
There is the possibility to reduce the time in which a minimal change is notified. 
The problem is to find out which node is "crazy" here.

The API does not provide good debugging.
The nodes already offer these.
Please stop the Nodered service and enable debugging by manually starting nodered and using 
`DEBUG=node-red-contrib-ads:*,node-ads|<your nodered start>`

Please take a look at who often sends or receives here.

I do not want to rule out an error. But I do not currently have this problem. There was a CPU problem with the reconnect. But that is solved.

hbTecAG, 2019-03-26:
Hello

You can close this one. i Have a problem with one variable in the PLC, that send every PLC Cycle. When i Force this variable in TwinCAT, everything works...

closed 2019-04-27 by PLCHome