PLCHome/node-red-contrib-ads #18: ADS notification not working after PLC has been offline (same as #14)

hbTecAG, 2019-03-28:
Hello

I have the same problem as issue #14. I have wrote in #14 but i don't know, that you see this beceause its closed.. 

Have the same problem. After a reboot of the CX (5020) the Notification Node does not work anymore ..
What i can do?
Even if the variables change in the PLC, the changes do not get into the Node-RED

Please mark as open Issue

Version: 1.1.20 (Updated today)

Log:
6 Mar 11:46:25 - [info] Starting flows 26 Mar 11:46:25 - [info] Started flows 26 Mar 11:47:23 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: read ECONNRESET 26 Mar 11:47:25 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: connect ECONNREFUSED 10.1.121.1:48898 26 Mar 11:49:15 - [info] Stopping flows 26 Mar 11:49:16 - [info] Stopped flows 26 Mar 11:49:16 - [info] Starting flows 26 Mar 11:49:16 - [info] Started flows 26 Mar 11:49:30 - [info] Stopping flows 26 Mar 11:49:31 - [info] Stopped flows 26 Mar 11:49:31 - [info] Starting flows 26 Mar 11:49:32 - [info] Started flows 26 Mar 11:50:06 - [info] Stopping flows 26 Mar 11:50:07 - [info] Stopped flows 26 Mar 11:50:07 - [info] Starting flows 26 Mar 11:50:08 - [info] Started flows 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 [... cut ...] 26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:26 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: target port not found 26 Mar 11:50:27 - [error] [ads-connection:64632e0d.6f1a5] Error on connect: check target NetId or routing 26 Mar 11:50:50 - [error] [ads-connection:64632e0d.6f1a5] Error ADS: Error: connect ECONNREFUSED 10.1.121.1:48898

PLCHome, 2019-03-28:
I thought I had solved the bug. 
Sorry for that. 
I'm still in holidays when I'm back I Booking on this more closely.
I have an idea why that is. 
There are some gaps in the feedback of connection aborts to the ads-connection

hbTecAG, 2019-03-28:
Hello

No problem, enjoy your holidays ;-)

Do you need more information? Or is that log enough?

The status Node always shows, the connection is ok and the PLC is RUN. Only the Notification Node will not work anymore. After a Deploy, you will be back.

If you need more information, get in touch!

hbTecAG, 2019-04-11:
Hey there

Do you have any idea to solve the problem?

PLCHome, 2019-04-12:
If I stop the PLC then the ADS Connection goes into an error mode:

```
node-red-contrib-ads:adsSystemNode onData: node.id a4f17781.928448 node.type ADS System msg { payload:
 { connectState: -1,
   connectStateText: 'ERROR',
   adsState: 0,
   adsStateText: 'INVALID',
   majorVersion: 2,
   minorVersion: 11,
   versionBuild: 2605,
   version: '2.11.2605',
   deviceName: 'TCatPlcCtrl',
   symTab: 89 },
_msgid: '71c3e326.67d84c' } +543ms
node-red-contrib-ads:adsConnectionNode setSystemStatus: enter +4ms
node-red-contrib-ads:adsConnectionNode setSystemStatus: a4f17781.928448 ADS System { fill: 'red', shape: 'ring', text: 'ERROR' } +2ms
node-red-contrib-ads:adsConnectionNode startTimer: Timeout +20s

```

When restarting the connection is first established while the PLC seems to be still inactive. The notification is already registered. That probably does not work. I have to look how I can do that the registration happens only at run.

```
node-red-contrib-ads:adsConnectionNode internalRestart: enter +6ms
node-red-contrib-ads:adsConnectionNode internalSetConnectState: new state:3 old state:-1 +1ms
node-red-contrib-ads:adsConnectionNode internalSetConnectState: set state:3 +2ms
node-red-contrib-ads:adsConnectionNode internalSetAdsState: set State: 0 +1ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: enter +1ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: enter +1ms
node-red-contrib-ads:adsConnectionNode internalRestart: done no adsClient +1ms
node-red-contrib-ads:adsConnectionNode internalSetConnectState: new state:0 old state:3 +9ms
node-red-contrib-ads:adsConnectionNode internalSetConnectState: set state:0 +2ms
node-red-contrib-ads:adsConnectionNode internalSetAdsState: set State: 0 +1ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: enter +2ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: enter +1ms
node-red-contrib-ads:adsConnectionNode internalSetConnectState: new state:1 old state:0 +2ms
node-red-contrib-ads:adsConnectionNode internalSetConnectState: set state:1 +7ms
node-red-contrib-ads:adsConnectionNode internalSetAdsState: set State: 0 +1ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: enter +1ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: enter +2ms
node-red-contrib-ads:adsConnectionNode connect: { host: '192.168.2.60',
amsNetIdTarget: '192.168.2.60.1.1',
amsNetIdSource: '192.168.2.62.1.1',
port: 48898,
amsPortSource: 32905,
amsPortTarget: 801,
timeout: 500 } +1ms
node-red-contrib-ads:adsConnectionNode removeClient: Client: false symbolsCache: false datatypsCache: false +2ms
node-red-contrib-ads:adsConnectionNode connect: readDeviceInfo: { majorVersion: 2,
minorVersion: 11,
versionBuild: 2605,
deviceName: 'TCatPlcCtrl' } +13ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: enter +2ms
node-red-contrib-ads:adsConnectionNode internalSubscribeLiveTick: +8ms
node-red-contrib-ads:adsConnectionNode internalSubscribeLiveTick: { indexGroup: 61696,
indexOffset: 0,
transmissionMode: 3,
cycleTime: 1000,
bytelength: { length: 2, name: 'WORD' } } +1ms
node-red-contrib-ads:adsConnectionNode internalSubscribeLiveTick: +2ms
node-red-contrib-ads:adsConnectionNode internalSubscribeSymTab: { indexGroup: 61448,
indexOffset: 0,
transmissionMode: 4,
bytelength: { length: 1, name: 'BYTE' },
start: true } +1ms
node-red-contrib-ads:adsConnectionNode internalSubscribe: ae4557cc.adb708 ADS Notification +2ms
node-red-contrib-ads:adsConnectionNode internalSetConnectState: new state:2 old state:1 +1ms
node-red-contrib-ads:adsConnectionNode internalSetConnectState: set state:2 +2ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: enter +9ms
[... 33 lines cut ...]
node-red-contrib-ads:adsConnectionNode setSystemStatus: a4f17781.928448 ADS System { fill: 'grey', shape: 'dot', text: 'INVALID' } +1ms
node-red-contrib-ads:adsConnectionNode internalSetAdsState: set State: 5 +906ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: enter +6ms
node-red-contrib-ads:adsConnectionNode internalSystemUpdate: call a4f17781.928448 ADS System +51ms
node-red-contrib-ads:adsConnectionNode systemUpdate: enter a4f17781.928448 ADS System +5ms
node-red-contrib-ads:adsSystemNode onData: node.id a4f17781.928448 node.type ADS System msg { payload:
 { connectState: 2,
   connectStateText: 'CONNECTED',
   adsState: 5,
   adsStateText: 'RUN',
   majorVersion: 2,
   minorVersion: 11,
   versionBuild: 2605,
   version: '2.11.2605',
   deviceName: 'TCatPlcCtrl',
   symTab: 90 },
_msgid: 'c976eab9.3a0318' } +972ms
node-red-contrib-ads:adsConnectionNode setSystemStatus: enter +4ms
node-red-contrib-ads:adsConnectionNode setSystemStatus: a4f17781.928448 ADS System { fill: 'green', shape: 'dot', text: 'RUN' } +1ms

```

High CPU load has not occurred with me. 
Also, node red tries a reconnect every 20 seconds.
Maybe I will do something different too.

PLCHome, 2019-04-13:
corrected in 1.1.21

closed 2019-04-13 by PLCHome