PLCHome/node-red-contrib-ads #19: String output: The "size" argument must be of type number. Received type string

brechtgijsens, 2019-04-19:
Hello, when I try to send a string to my PLC I get an error:
The "size" argument must be of type number. Received type string.

I made a textbox with mode "text input" this outputs a string. This string is then passed in the output node. I defined it as a string and gave it size: 80.
[image]

Then after putting in a string in the textbox, Node-RED crashes.
[image]

Reading strings works perfect, just like everything else.

PLCHome, 2019-04-27:
Oh sorry:

@dkleber89 changes in the API "new Buffer" in "Buffer.alloc"
old:  var data = new Buffer(length)
new:  var data = Buffer.alloc(length)

But on row 1623 it must be 
old:    var stringbuf = new Buffer(val.toString().slice(0,convert.length-1) + '\0', 'binary')
wrong:   var stringbuf = Buffer.alloc(val.toString().slice(0,convert.length-1) + '\0', 'binary')
new:   var stringbuf = Buffer.from(val.toString().slice(0,convert.length-1) + '\0', 'binary')

closed 2019-04-27 by PLCHome

PLCHome, 2019-04-27:
solved in version 1.1.22