PLCHome/node-red-contrib-ads #20: Tutorial to config Beckhoff PLC

Tsepatek, 2019-05-02:
Hi,
I am a newbie in Node-Red and also with Beckhoff system.
Could someone give a tutorial for dummies how to configure Beckhoff to work in Node-Red?

Where in Beckhoff System Manager I can find the right values to
Host, Target NetId, Source NetId, Port, Target Port, Source Port?
My TwinCat is version 2.
Can I configure Node-Red to connect directly to my Beckhoff PC (Windows CE)?

Please help me.

PLCHome, 2019-05-02:
First tip for newbies: Do not run Twincat and Node Red ADS on the same machine. That makes problems, best to install Nodered in a VM, on a second PC or on a Raspberry.

> Can I configure Node-Red to connect directly to my Beckhoff PC (Windows CE)?

Yes that's exactly what it's for.

> Where in Beckhoff System Manager I can find the right values to
Host, Target NetId, Source NetId, Port, Target Port, Source Port?

Actually everything is in the readme md.

Open your project in the system manager and enter the IP of the PC running on the Nodered (in my case the 192.168.2.10). For AMS-Net-ID use the IP address + .1.1. You need to update and reboot
[image]

Hier you can find your PLC NetID

[image]

Open your node red in a browser

Insert the example into node-red (Menu Import clipboard):
````
[{"id":"b0a9c705.017d08","type":"ADS Symbols","z":"524f011a.f0bce8","name":"","datasource":"","data":"SYMBOLES","x":370,"y":940,"wires":[["f9278f67.b8ba2"]]},{"id":"f9278f67.b8ba2","type":"ui_template","z":"524f011a.f0bce8","group":"aa9409af.67136","name":"","order":0,"width":"26","height":"6","format":"<style>\ntable {\n    color: #333;\n    font-family: Helvetica, Arial, sans-serif;\n    width: 100%;\n    border-collapse: collapse;\n    border-spacing: 0;\n}\ntd, th {\n    border: 1px solid transparent;\n    height: 30px;\n    transition: all 0.3s;\n}\nth {\n    background: #DFDFDF;\n    /* Darken header a bit */\n    font-weight: bold;\n    text-align: left;\n}\ntd {\n    background: #FAFAFA;\n    text-align: left;\n}\ntr:nth-child(even) td {\n    background: #F1F1F1;\n}\ntr:nth-child(odd) td {\n    background: #FEFEFE;\n}\ntr td:hover {\n    background: #666;\n    color: #FFF;\n}\n</style>\n\n<div>\n<table>\n  <tr>\n    <th>Index Group</th>\n    <th>Index Offset</th>\n    <th>Name</th>\n    <th>Type</th>\n    <th>Comment</th>\n  </tr>\n  <tr ng-repeat=\"obj in msg.payload\">\n    <td>{{ obj.indexGroup }}</td>\n    <td>{{ obj.indexOffset }}</td>\n    <td>{{ obj.name }}</td>\n    <td>{{ obj.type }}</td>\n    <td>{{ obj.comment }}</td>\n  </tr>\n</table>\n</div>","storeOutMessages":false,"fwdInMessages":false,"templateScope":"local","x":600,"y":940,"wires":[[]]},{"id":"cf2f5485.8f3268","type":"ADS Symbols","z":"524f011a.f0bce8","name":"","datasource":"","data":"TYPES","x":360,"y":1020,"wires":[["803b7b42.ce51b"]]},{"id":"803b7b42.ce51b","type":"ui_template","z":"524f011a.f0bce8","group":"f880e964.b5012","name":"","order":2,"width":"26","height":"10","format":"<style>\ntable {\n    color: #333;\n    font-family: Helvetica, Arial, sans-serif;\n    width: 100%;\n    border-collapse: collapse;\n    border-spacing: 0;\n}\ntd, th {\n    border: 1px solid transparent;\n    height: 30px;\n    transition: all 0.3s;\n}\nth {\n    background: #DFDFDF;\n    font-weight: bold;\n    text-align: left;\n}\ntd {\n    background: #FAFAFA;\n    text-align: left;\n}\ntr:nth-child(even) td {\n    background: #F1F1F1;\n}\ntr:nth-child(odd) td {\n    background: #FEFEFE;\n}\ntr td:hover {\n    background: #666;\n    color: #FFF;\n}\n</style>\n\n<div>\n<table>\n  <tr>\n    <th>Version</th>\n    <th>Size</th>\n    <th>Data Type</th>\n    <th>Name</th>\n    <th>Type</th>\n    <th>Comment</th>\n    <th>Array Dim</th>\n    <th>Sub Types</th>\n  </tr>\n  <tr ng-repeat=\"obj in msg.payload\">\n    <td>{{ obj.version }}</td>\n    <td>{{ obj.size }}</td>\n    <td>{{ obj.dataType }}</td>\n    <td>{{ obj.name }}</td>\n    <td>{{ obj.type }}</td>\n    <td>{{ obj.comment }}</td>\n    <td>\n       <table ng-show=\"obj.array\">\n          <tr>\n            <th>Low Bound</th>\n            <th>Elements</th>\n          </tr>\n          <tr ng-repeat=\"sobj in obj.array\">\n            <td>{{ sobj.lBound }}</td>\n            <td>{{ sobj.elements }}</td>\n          </tr>\n       </table>\n     </td>       \n    <td>\n       <table ng-show=\"obj.datatyps\">\n          <tr>\n            <th>Index</th>\n            <th>Version</th>\n            <th>Size</th>\n            <th>Data Type</th>\n            <th>Offset</th>\n            <th>Name</th>\n            <th>Type</th>\n            <th>Comment</th>\n          </tr>\n          <tr ng-repeat=\"sobj in obj.datatyps\">\n            <td>{{ sobj.index }}</td>\n            <td>{{ sobj.version }}</td>\n            <td>{{ sobj.size }}</td>\n            <td>{{ sobj.dataType }}</td>\n            <td>{{ sobj.offs }}</td>\n            <td>{{ sobj.name }}</td>\n            <td>{{ sobj.type }}</td>\n            <td>{{ sobj.comment }}</td>\n          </tr>\n       </table>\n     </td>       \n  </tr>\n</table>\n</div>","storeOutMessages":false,"fwdInMessages":false,"templateScope":"local","x":600,"y":1020,"wires":[[]]},{"id":"6dadd36d.9d2334","type":"ui_text_input","z":"524f011a.f0bce8","name":"","label":"Search","group":"f880e964.b5012","order":1,"width":0,"height":0,"passthru":false,"mode":"text","delay":"0","topic":"","x":140,"y":1020,"wires":[["cf2f5485.8f3268"]]},{"id":"8dfe926d.20cbb","type":"ui_text_input","z":"524f011a.f0bce8","name":"","label":"Search","group":"aa9409af.67136","order":1,"width":"0","height":"0","passthru":false,"mode":"text","delay":"0","topic":"","x":140,"y":940,"wires":[["b0a9c705.017d08"]]},{"id":"aa9409af.67136","type":"ui_group","z":"","name":"Symboles","tab":"65f5295f.68b3c","order":1,"disp":true,"width":"26","collapse":true},{"id":"f880e964.b5012","type":"ui_group","z":"","name":"Types","tab":"65f5295f.68b3c","order":2,"disp":true,"width":"26","collapse":true},{"id":"65f5295f.68b3c","type":"ui_tab","z":"","name":"inspect PLC","icon":"fingerprint","order":2}]
````

Open Propery of the Symbols and Types. On one enter and on one select the PLC.

[image]

[image]

Host: IP of the PLC
Target NetID: NetID of the PLC
Source NeiID:Your NetID

Check that there is only on configuration node:
[image]

PLCHome, 2019-05-02:
If you want to support this project, feel free to write a tutorial with your experiences and make a merge request.

Tsepatek, 2019-05-03:
Hi,
Thank you. Now I am able to connect to my PLC.
I will document all I have done and contribute the documentation later.

closed 2019-05-03 by PLCHome

Tsepatek, 2019-05-09:
Hi,
My project continues. I am now able to read for example Lamp status (0 or 1) from the Output variable .arrLampProcessOutputData[17].bdata in Beckhoff system. The problem is that Beckhoff does not accept if I try to set the .arrLampProcessOutputData[17].bdata value to 0 or 1. How should I configure for example the switch (dasboard item) to send on or off command to Beckhoff? Should I use Outputs or what? Could someone give an example?
Help needed again.

365Andy, 2020-06-17:
> First tip for newbies: Do not run Twincat and Node Red ADS on the same machine. That makes problems, best to install Nodered in a VM, on a second PC or on a Raspberry.
>
>
> Yes that's exactly what it's [...]

Hi, How would the Target NetId and the Source NetId and the rest of the properties configured if the TC3.1 XAR is on the same PC / Target device as where Node-Red is running. XAR and Node-Red on the same device?

Tanks in advance for the help.

panostaja, 2021-08-07:
> Hi,
> My project continues. I am now able to read for example Lamp status (0 or 1) from the Output variable .arrLampProcessOutputData[17].bdata in Beckhoff system. The problem is that Beckhoff does not accept if I try to set the .arrLampProcessOutputData[17].bdata value to 0 or 1. How should I configure for example the switch (dasboard item) to send on or off command to Beckhoff? Should I use Inputs or what? Could someone give an example?
> Help needed again.

Hi!

Waking up this topic, because was looking for the same information and couldn't find any solution anywhere.  After couple of days of going through "manual" and tweaking, found the solution anyway: Write control value 0-100 (LREAL) to .arrLampCommands[xx].lrSetControlValueDirect and then activate the value (which was actually saved to lrResettingValue) by writing 1 (BOOL) to .arrLampCommands[xx].bSetControlValueDirect. Be aware that the array values are not the same (so 17 most likely isn't working and you need to dig the array number up from TwinCat System Manager terminal channel "linked into" field.