PLCHome/node-red-contrib-ads #21: Unable to set values to Beckhoff PLC. Reading the values is ok.

Tsepatek, 2019-05-09:
Hi,
My project continues. I am now able to read for example Lamp status (0 or 1) from the Output variable .arrLampProcessOutputData[17].bdata in Beckhoff system. The problem is that Beckhoff does not accept if I try to set the .arrLampProcessOutputData[17].bdata value to 0 or 1. How should I configure for example the switch (dasboard item) to send on or off command to Beckhoff? Should I use Inputs or symbols or what? I was able to list all the symbols in Beckhoff system.
Could someone give an example?
Help needed again.

PLCHome, 2019-05-09:
- ADS Notification responds to data changes within the PLC. 
- ADS IN fetches the value from the PLC when written to the input of the ADS IN. The node is not needed so often.
- ADS OUT writes the input to the PLC.
- ADS Systen shows the connection and system state to the PLC
- ADS Symbols is to explore the PLC

This is an example on the Node-Red
[image]

```
[{"id":"d0b01be6.4ab07","type":"debug","z":"7a7b8638.9b31d8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":880,"y":1240,"wires":[]},{"id":"39a19e59.c4b172","type":"ADS Notification","z":"7a7b8638.9b31d8","name":"","datasource":"","varName":".arrLampProcessOutputData[17].bdata ","varTyp":"BOOL","varSize":"","timezone":"UNCHANGED","transmissionMode":"ONCHANGE","maxDelay":0,"cycleTime":10,"property":"payload","topic":"","x":610,"y":1240,"wires":[["d0b01be6.4ab07"]]},{"id":"fa63dbd4.bea1e","type":"ADS Output","z":"7a7b8638.9b31d8","name":"","datasource":"","varName":".arrLampProcessOutputData[17].bdata ","varTyp":"BOOL","outValue":"payload","varSize":"","timezone":"UNCHANGED","topic":"","x":620,"y":1300,"wires":[]},{"id":"c38e690a.ee72a8","type":"inject","z":"7a7b8638.9b31d8","name":"","topic":"","payload":"0","payloadType":"num","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":310,"y":1280,"wires":[["fa63dbd4.bea1e"]]},{"id":"d63b2c4f.a96d58","type":"inject","z":"7a7b8638.9b31d8","name":"","topic":"","payload":"1","payloadType":"num","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":310,"y":1320,"wires":[["fa63dbd4.bea1e"]]}]
```
This is an example on the Dashboard
[image]
[image]

```
[{"id":"39a19e59.c4b172","type":"ADS Notification","z":"7a7b8638.9b31d8","name":"","datasource":"","varName":".arrLampProcessOutputData[17].bdata ","varTyp":"BOOL","varSize":"","timezone":"UNCHANGED","transmissionMode":"ONCHANGE","maxDelay":0,"cycleTime":10,"property":"payload","topic":"","x":290,"y":1180,"wires":[["eda78d15.58983"]]},{"id":"fa63dbd4.bea1e","type":"ADS Output","z":"7a7b8638.9b31d8","name":"","datasource":"","varName":".arrLampProcessOutputData[17].bdata ","varTyp":"BOOL","outValue":"payload","varSize":"","timezone":"UNCHANGED","topic":"","x":860,"y":1180,"wires":[]},{"id":"eda78d15.58983","type":"ui_switch","z":"7a7b8638.9b31d8","name":"","label":"Test","tooltip":"","group":"af7385c5.b1f718","order":1,"width":"6","height":"1","passthru":false,"decouple":"false","topic":"","style":"","onvalue":"true","onvalueType":"bool","onicon":"","oncolor":"","offvalue":"false","offvalueType":"bool","officon":"","offcolor":"","x":570,"y":1180,"wires":[["fa63dbd4.bea1e"]]},{"id":"af7385c5.b1f718","type":"ui_group","z":"","name":"Default","tab":"2bb65a3a.2f71a6","disp":true,"width":"6","collapse":false},{"id":"2bb65a3a.2f71a6","type":"ui_tab","z":"","name":"Home","icon":"dashboard","order":1}]
```

Of the nodes still the Datasource must be connected.

Tsepatek, 2019-05-11:
Thank you for the quick answer. I have tried both of those examples. I think the problem is with Beckhoff, because it does not accept value change to Outputs .arrLampProcessOutputData[17].bdata. I talked to my Beckhoff system support person and he told me that output data cannot be changed directly. Now I need to know which variable I need to use to get Outputs value .arrLampProcessOutputData[17].bdata to change. I can force the value in Beckhoff Manager software, but I cannot force the value by using Nodered ADS Output.

PLCHome, 2019-05-11:
There are some limitations, variables associated with outputs can not be written through ADS. It can also happen that variables within programs can not be written. 
I had the same problems before Node-Red with the ADS webservices. 
I have solved problems in Twincat by writing in the global variable as a cache and copy it at the end of the program to the outputs.

First, start small. No structures of types and no variables mapped to outputs.

closed 2019-05-12 by PLCHome