PLCHome/node-red-contrib-ads #22: Can't Update

hbTecAG, 2019-05-24:
Hello

When i want to Update to latest, npm installs automaticly 1.1.16..

also when i want to install new with `npm i node-red-contrib-ads` the answer ist: 
`npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EAI_AGAIN: request to https://registry.npmjs.org/node-red-contrib-ads failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org:443
npm WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.
npm WARN node-red-project@0.0.1 No repository field.
npm WARN node-red-project@0.0.1 No license field.

+ node-red-contrib-ads@1.1.16
updated 1 package in 131.639s`

What i do wrong?

PLCHome, 2019-05-24:
To update use the update not the install: `npm update -g node-red-contrib-ads`

Or uninstall `npm uninstall node-red-contrib-ads` the package and reinstall it over node-red there ist  alson an update button in the "palette".

[image]

hbTecAG, 2019-05-24:
I know the Button in the Palette.. But it doesn't works.. 

When i make: `npm update -g node-red-contrib-ads` it comes no error, but the version is still the same. 
When i uninstall and install again, also the Version 1.1.16 is installed... ?

I think, it is because of the last release in github is the 1.1.16. Does npm get this Version by updating?

[image]

PLCHome, 2019-05-24:
I'm not so good with the npm. 
The version management of GitHub has nothing to do with NPM storage. 
I assume that the installation has been changed and npm can not retrieve. 
I use the update with npm in my productive environment by myself.

The error is in your installation not in the published software.

closed 2019-05-27 by hbTecAG