PLCHome/node-red-contrib-ads #25: Error: "Ads Register Notification timeout"

hbTecAG, 2020-03-09:
Hi there

I have following problem: 

i have set up a Project, that have around 350 ads nodes. (ADS Notification and ADS Outputs). -->i mean, thats under the Limit of the readme.. 

When i restart the PLC, everything is ok. But from the 1st or 2nd Deploy in Node-RED i have following Errors: (@every Deploy and peridically)

- "Ads Register Notification timeout". 
Sometimes followed by: 
"Ads write timeout" and / or  "This socket has been ended by the other party"

I have tried to set the timeout in the ads config node to 2000ms with the same result. 

Node-RED: v1.0.3
Nodejs: 11.9.0
ADS: 1.1.22

My Node-RED is Running in a Docker Container on a Raspberry Pi and the PLC is Running on a CX5120 in the same Network. 

It seems as the connection doesn't close correctly or something like that. Do you have an idea what can be wrong?

PLCHome, 2020-03-09:
I also ran node red in a Docker on a raspi.

I know the problem. I work around it by deploying changed nodes and not all nodes.

I have the suspicion that this is due to the deregistration of the notifications from the PLC.

I built an end button into the flow as a work around. Supervisored then restarts nodered.

hbTecAG, 2020-03-10:
Thanks for your quick reply.

Can you send me an example of this "end button"? (Export flow)

Can the problem be solved? It is not a permanent solution to always restart Node-RED ...

PLCHome, 2020-03-11:
> Can the problem be solved? It is not a permanent solution to always restart Node-RED ...
I have less time at the moment, the API would have to be rebuilt or rewritten

> Can you send me an example of this "end button"? (Export flow)
For sure, but you have to take care of the restart in the docker

[{"id":"7b6d32e4.96402c","type":"inject","z":"8722e696.cdc288","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":300,"y":120,"wires":[["c05daa66.646f18"]]},{"id":"25e95aa0.fcb066","type":"ui_button","z":"8722e696.cdc288","name":"","group":"4b2ecacd.de6cf4","order":2,"width":0,"height":0,"passthru":false,"label":"Restart Nodered","tooltip":"","color":"","bgcolor":"","icon":"","payload":"true","payloadType":"bool","topic":"","x":270,"y":180,"wires":[["c05daa66.646f18"]]},{"id":"c05daa66.646f18","type":"exec","z":"8722e696.cdc288","command":"pkill node-red","addpay":false,"append":"","useSpawn":"true","timer":"","oldrc":false,"name":"","x":540,"y":120,"wires":[[],[],[]]},{"id":"4b2ecacd.de6cf4","type":"ui_group","z":"","name":"Nodered","tab":"6938039d.a0584c","order":1,"disp":true,"width":"6","collapse":false},{"id":"6938039d.a0584c","type":"ui_tab","z":"","name":"Admin","icon":"dashboard","order":7,"disabled":false,"hidden":false}]

hbTecAG, 2020-03-25:
Hello

I hope you are well?

Is it possible to tackle this problem? Your solution with restarting is not really practical. Do you see a time frame where you could look at it?

hbTecAG, 2020-05-06:
Why did you close this? The issue is still there...
don't you want to fix that?

pyrler, 2020-08-15:
same problem here. any solution?

PLCHome, 2020-08-16:
I Fixed something in 1.1.24

hbTecAG, 2020-08-26:
Hello

i have still the same Problem:

1.1.24
Node-RED 1.1.3
NodeJS: 12.16.1

`
26 Aug 19:00:42 - [info] Started flows
26 Aug 19:00:43 - [error] [ADS Notification:13402638.1aae9a] Ads Register Notification '.GMdsp1_Preset3SelectUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:bbb8314e.4d911] Ads Register Notification '.GMdsp1_Preset4SelectUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:5755da07.aa5b94] Ads Register Notification '.GMStatusAnlage_LED' timeout
26 Aug 19:00:43 - [error] [ADS Notification:5243cf10.35393] Ads Register Notification '.GMdsp1_Mic1EnableUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:d85e503f.f800d] Ads Register Notification '.GMdsp1_Mic1plus6dBUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:f70fd1be.0b47a] Ads Register Notification '.GMdsp1_Mic1minus3dBUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:18ccefb4.131c9] Ads Register Notification '.GMdsp1_Mic1minus6dBUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:fbc0dd81.9cb0c] Ads Register Notification '.GMdsp1_Mic2EnableUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:ac624b61.f72eb8] Ads Register Notification '.GMdsp1_Mic3plus6dBUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:91dad737.eead68] Ads Register Notification '.GMdsp1_Mic3plus3dBUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:69140439.bdf41c] Ads Register Notification '.GMdsp1_Mic3plus0dBUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:c60659f2.606d98] Ads Register Notification '.GMdsp1_Mic3minus3dBUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:a1834e60.c399e] Ads Register Notification '.GMdsp1_Mic3minus6dBUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:43e85286.734ebc] Ads Register Notification '.GMdsp1_Musik1EnableUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:62428ef0.f7656] Ads Register Notification '.GMdsp1_Musik1SliderWertUi' timeout
26 Aug 19:00:43 - [error] [ADS Notification:7df2da38.587614] Ads Register Notification '.GMdsp1_Musik2EnableUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:bb269882.f88528] Ads Register Notification '.GMdsp1_Musik2SliderWertUi' timeout
26 Aug 19:00:43 - [error] [ADS Notification:3698393a.4e24a6] Ads Register Notification '.oMBeamerLED' timeout
26 Aug 19:00:43 - [error] [ADS Notification:dd8bc572.99d2c8] Ads Register Notification '.GMBe1_Quelle1Uo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:db3966e5.ab1188] Ads Register Notification '.GMBe1_Quelle2Uo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:143e5479.dc6d7c] Ads Register Notification '.GMBe1_MuteUo' timeout
26 Aug 19:00:43 - [error] [ADS Notification:38d1697f.49a776] Ads Register Notification '.GMBe1_LampeLebenszeitUo' timeout
26 Aug 19:00:43 - [error] [ADS Output:1e5915a0.2d9b5a] Ads write '.gMemHeizKalender[0][1].valid' timeout
26 Aug 19:00:43 - [error] [ADS Output:6eb7a641.453bb8] Ads write '.oLAuss' timeout
`

PLCHome, 2020-08-26:
what is your timeout in config node?

hbTecAG, 2020-08-27:
Hi

Thanks for your reply

timeout in confignode: 500
[image]

I have set it up to 1000, after the 3nd Deploy i have the same problem..
The next problem, when i have a timeout at start, i dont's have communication at all time if i klick some buttons.. nothing matters on the sps. the Conncetion is alive (ADS System shows RUN) but nothing happens..

PLCHome, 2020-08-27:
> My Node-RED is Running in a Docker Container on a Raspberry Pi 
Same with me. 
I don't have the problem anymore. Why, I have no idea. 
Can you activate the debugging, also cyclically? I don't know now whether you are running under supervisor or run in the container - that makes a difference when it comes to activation. With supervisor you can simply exit nodeRed and set DEBUG and start it manually.
The log is getting really big. But I don't know how else to find it.

hbTecAG, 2020-08-27:
Hi

i don't think, that this can help you :-)
i can't see mutch more than the normal loglevel (info) and after a log with level trace

Let me know if you use more or other infos

[debug log.txt](https://github.com/PLCHome/node-red-contrib-ads/files/5136472/debug.log.txt)

[trace log.txt](https://github.com/PLCHome/node-red-contrib-ads/files/5136470/trace.log.txt)

PLCHome, 2020-08-27:
Yes, that's actually not what I meant. 

I thought I had to describe it. But it is only available for the Velux adapter.

To record all function calls for ads nodes, the environment variable DEBUG must be set:
`DEBUG=node-red-contrib-ads:*`

And without the timer:
`DEBUG=node-red-contrib-ads:*,-node-red-contrib-ads:adsConnectionNode:Cyclic`

hbTecAG, 2020-08-27:
Hi,

i have seen, the Pi is >100% CPU:
[image]

If i edit the config node's IP from ads-config Node to a IP that doesn't exist, everything is ok. If i edit the IP to the SPS the CPU is >100%. 

I will try to debug more..
[2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json3.txt](https://github.com/PLCHome/node-red-contrib-ads/files/5136983/2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json3.txt)
[2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json4.txt](https://github.com/PLCHome/node-red-contrib-ads/files/5136984/2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json4.txt)
[2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json.txt](https://github.com/PLCHome/node-red-contrib-ads/files/5136985/2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json.txt)
[2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json1.txt](https://github.com/PLCHome/node-red-contrib-ads/files/5136986/2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json1.txt)
[2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json2.txt](https://github.com/PLCHome/node-red-contrib-ads/files/5136987/2f488d657ad1584d1e4b4386c83b6822282bec7331fe7b733290a0656f50dda9-json2.txt)

hbTecAG, 2020-08-27:
Here a Logfile from Beginning

[8d90b70f0a660afbb5eee2826b3c9a347bc63dc408a26ae46c9a89447b2fe088-json.log2.txt](https://github.com/PLCHome/node-red-contrib-ads/files/5137033/8d90b70f0a660afbb5eee2826b3c9a347bc63dc408a26ae46c9a89447b2fe088-json.log2.txt)

PLCHome, 2020-08-27:
I do not see any connection errors or registration errors, only that the notification is set briefly:

.GK_Befehle[49].Name
.GK_Befehle[36].Name
...

Can you send me a hard copy of the nodes?

hbTecAG, 2020-08-28:
`[
    {
        "id": "aadd8141.feb8b",
        "type": "ADS Notification",
        "z": "e1184e4a.fa182",
        "name": "",
        "datasource": "d036abf9.50d8c8",
        "varName": ".GK_Befehle[50].Name",
        "varTyp": "STRING",
        "varSize": "",
        "timezone": "UNCHANGED",
        "transmissionMode": "ONCHANGE",
        "maxDelay": 0,
        "cycleTime": 10,
        "property": "payload",
        "topic": "",
        "x": 140,
        "y": 840,
        "wires": [
            [
                "d62b81db.3240a"
            ]
        ]
    },
    {
        "id": "f6876c15.ed686",
        "type": "ADS Notification",
        "z": "e1184e4a.fa182",
        "name": "",
        "datasource": "d036abf9.50d8c8",
        "varName": ".GK_Befehle[49].Name",
        "varTyp": "STRING",
        "varSize": "",
        "varLowIndex": "",
        "varHighIndex": "",
        "isArray": false,
        "timezone": "UNCHANGED",
        "transmissionMode": "ONCHANGE",
        "maxDelay": 0,
        "cycleTime": 10,
        "property": "payload",
        "topic": "",
        "x": 140,
        "y": 1120,
        "wires": [
            [
                "74b2dda4.fa5cf4"
            ]
        ]
    },
    {
        "id": "c38c388f.533f88",
        "type": "ADS Notification",
        "z": "e1184e4a.fa182",
        "name": "",
        "datasource": "d036abf9.50d8c8",
        "varName": ".GK_Befehle[34].Name",
        "varTyp": "STRING",
        "varSize": "",
        "timezone": "UNCHANGED",
        "transmissionMode": "ONCHANGE",
        "maxDelay": 0,
        "cycleTime": 10,
        "property": "payload",
        "topic": "",
        "x": 160,
        "y": 1720,
        "wires": [
            [
                "8e9c73f6.5d11"
            ]
        ]
    },
    {
        "id": "381e64ac.431b7c",
        "type": "ADS Notification",
        "z": "e1184e4a.fa182",
        "name": "",
        "datasource": "d036abf9.50d8c8",
        "varName": ".GK_Befehle[35].Name",
        "varTyp": "STRING",
        "varSize": "",
        "timezone": "UNCHANGED",
        "transmissionMode": "ONCHANGE",
        "maxDelay": 0,
        "cycleTime": 10,
        "property": "payload",
        "topic": "",
        "x": 160,
        "y": 2000,
        "wires": [
            [
                "8a8fac3a.79219"
            ]
        ]
    },
    {
        "id": "32e0deaa.9684f2",
        "type": "ADS Notification",
        "z": "e1184e4a.fa182",
        "name": "",
        "datasource": "d036abf9.50d8c8",
        "varName": ".GK_Befehle[36].Name",
        "varTyp": "STRING",
        "varSize": "",
        "varLowIndex": "",
        "varHighIndex": "",
        "isArray": false,
        "timezone": "UNCHANGED",
        "transmissionMode": "ONCHANGE",
        "maxDelay": 0,
        "cycleTime": 10,
        "property": "payload",
        "topic": "",
        "x": 160,
        "y": 2280,
        "wires": [
            [
                "6785e4f2.756fcc"
            ]
        ]
    },
    {
        "id": "a6baa6af.7d8ea8",
        "type": "ADS Notification",
        "z": "e1184e4a.fa182",
        "name": "",
        "datasource": "d036abf9.50d8c8",
        "varName": ".GK_Befehle[37].Name",
        "varTyp": "STRING",
        "varSize": "",
        "timezone": "UNCHANGED",
        "transmissionMode": "ONCHANGE",
        "maxDelay": 0,
        "cycleTime": 10,
        "property": "payload",
        "topic": "",
        "x": 160,
        "y": 2580,
        "wires": [
            [
                "fc964970.1eb538"
            ]
        ]
    },
    {
        "id": "d036abf9.50d8c8",
        "type": "ads-connection",
        "z": "",
        "host": "10.1.121.3",
        "clientIP": "",
        "amsNetIdTarget": "10.1.121.1.1.1",
        "amsNetIdSource": "10.1.121.120.1.1",
        "port": "48898",
        "amsPortSource": "32905",
        "amsPortTarget": "801",
        "adsTimeout": "500"
    }
]`

PLCHome, 2020-08-28:
How long are the strings defined in the PLC?
I wonder what causes the PLC to transfer the data so often. 
And I don't see any change in the value.
A wrong size would be an idea if there is a value in the structure behind the string that is constantly changing, e.g. a timer would then trigger a 10ms transmission with this setting.
you can also try to set the cycle time to 1000 or 2000 for a test.
```
        "varTyp": "STRING",
        "varSize": "",                     <<== missing, so 80 characters I think
        "transmissionMode": "ONCHANGE",    <<== OK
        "maxDelay": 0,                     <<== OK
        "cycleTime": 10,                   <<== not faster than 10 ms

```

hbTecAG, 2020-08-28:
Hi,

The Variable is declared like this: 
(struct)
Name :		STRING[255] ;

I have set the size to 256Bytes  (CPU still high) and after i set the cyclic time to 1000 and the CPU is much better (<10%)

BUT: i have still much Timeouts at Deploy: (i edited all the ADS Notification Nodes with STRING's..)

`28 Aug 09:48:21 - [info] Flows gestartet
28 Aug 09:48:22 - [error] [ADS Notification:565702f7.8b0e5c] Ads Register Notification '.GHZuMe1TIst' timeout
28 Aug 09:48:22 - [error] [ADS Notification:b2797309.dd6ef] Ads Register Notification '.GHReg2Soll' timeout
28 Aug 09:48:22 - [error] [ADS Notification:69029f0a.4504d] Ads Register Notification '.GHReg2TIst' timeout
28 Aug 09:48:22 - [error] [ADS Notification:e851c483.fac338] Ads Register Notification '.GHReg2YOut' timeout
28 Aug 09:48:22 - [error] [ADS Notification:ebbc6ea3.3faf] Ads Register Notification '.GHZuMe2TIst' timeout
28 Aug 09:48:22 - [error] [ADS Notification:aadd8141.feb8b] Ads Register Notification '.GK_Befehle[50].Name' timeout
28 Aug 09:48:22 - [error] [ADS Notification:676e4157.5031c] Ads Register Notification '.GK_Div17BefehlUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:a0aa17.b757c5e8] Ads Register Notification '.GK_Glocken[1].Status' timeout
28 Aug 09:48:22 - [error] [ADS Notification:31ed69ac.3f9676] Ads Register Notification '.GK_Glocken[2].Status' timeout
28 Aug 09:48:22 - [error] [ADS Notification:8be930c9.9dd7a] Ads Register Notification '.GK_Glocken[3].Status' timeout
28 Aug 09:48:22 - [error] [ADS Notification:635f66d0.c5e8f8] Ads Register Notification '.GK_Glocken[4].Status' timeout
28 Aug 09:48:22 - [error] [ADS Notification:6ef7dbb8.cd18b4] Ads Register Notification '.GK_Glocken[5].Status' timeout
28 Aug 09:48:22 - [error] [ADS Notification:f6876c15.ed686] Ads Register Notification '.GK_Befehle[49].Name' timeout
28 Aug 09:48:22 - [error] [ADS Notification:18e939d5.604db6] Ads Register Notification '.GK_Div16BefehlUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:c38c388f.533f88] Ads Register Notification '.GK_Befehle[34].Name' timeout
28 Aug 09:48:22 - [error] [ADS Notification:bf54bedc.ef921] Ads Register Notification '.GK_Div01BefehlUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:381e64ac.431b7c] Ads Register Notification '.GK_Befehle[35].Name' timeout
28 Aug 09:48:22 - [error] [ADS Notification:8a7ceb13.b409a8] Ads Register Notification '.GK_Div02BefehlUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:32e0deaa.9684f2] Ads Register Notification '.GK_Befehle[36].Name' timeout
28 Aug 09:48:22 - [error] [ADS Notification:6ae0b889.181a98] Ads Register Notification '.GK_Div03BefehlUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:a6baa6af.7d8ea8] Ads Register Notification '.GK_Befehle[37].Name' timeout
28 Aug 09:48:22 - [error] [ADS Notification:ddd73916.d30058] Ads Register Notification '.GK_Div04BefehlUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:b73cef5f.9eb4f] Ads Register Notification '.GK_Vorschau[1]' timeout
28 Aug 09:48:22 - [error] [ADS Notification:939e9e7c.58ab7] Ads Register Notification '.GK_Vorschau[2]' timeout
28 Aug 09:48:22 - [error] [ADS Notification:32f7b4d9.de035c] Ads Register Notification '.GK_Vorschau[3]' timeout
28 Aug 09:48:22 - [error] [ADS Notification:14555748.c45da9] Ads Register Notification '.GK_GlockenDauer' timeout
28 Aug 09:48:22 - [error] [ADS Notification:4a10b253.fd525c] Ads Register Notification '.GMdsp1_Preset1SelectUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:d0b4c409.a4f3b8] Ads Register Notification '.GMdsp1_Preset2SelectUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:13402638.1aae9a] Ads Register Notification '.GMdsp1_Preset3SelectUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:bbb8314e.4d911] Ads Register Notification '.GMdsp1_Preset4SelectUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:5755da07.aa5b94] Ads Register Notification '.GMStatusAnlage_LED' timeout
28 Aug 09:48:22 - [error] [ADS Notification:5243cf10.35393] Ads Register Notification '.GMdsp1_Mic1EnableUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:d85e503f.f800d] Ads Register Notification '.GMdsp1_Mic1plus6dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:8be7dd5d.ffd67] Ads Register Notification '.GMdsp1_Mic1plus3dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:643b5015.84043] Ads Register Notification '.GMdsp1_Mic1plus0dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:f70fd1be.0b47a] Ads Register Notification '.GMdsp1_Mic1minus3dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:18ccefb4.131c9] Ads Register Notification '.GMdsp1_Mic1minus6dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:fbc0dd81.9cb0c] Ads Register Notification '.GMdsp1_Mic2EnableUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:c5c9b193.6a3d2] Ads Register Notification '.GMdsp1_Mic2plus6dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:575c2fcc.3ca48] Ads Register Notification '.GMdsp1_Mic2plus3dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:a81b65ee.ba0cd8] Ads Register Notification '.GMdsp1_Mic2plus0dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:2548e41f.3887dc] Ads Register Notification '.GMdsp1_Mic2minus3dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:e17c8e83.21b43] Ads Register Notification '.GMdsp1_Mic2minus6dBUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:4b5afa10.e1ce44] Ads Register Notification '.GMdsp1_Mic3EnableUo' timeout
[... cut ...]
28 Aug 09:48:22 - [error] [ADS Notification:dd8bc572.99d2c8] Ads Register Notification '.GMBe1_Quelle1Uo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:db3966e5.ab1188] Ads Register Notification '.GMBe1_Quelle2Uo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:143e5479.dc6d7c] Ads Register Notification '.GMBe1_MuteUo' timeout
28 Aug 09:48:22 - [error] [ADS Notification:38d1697f.49a776] Ads Register Notification '.GMBe1_LampeLebenszeitUo' timeout
28 Aug 09:48:23 - [error] [ADS Output:6eb7a641.453bb8] Ads write '.oLAuss' timeout
`

hbTecAG, 2020-08-28:
Here a Logfile completly from a new container till the timeouts recieved. i have deleted all nodes with string variables (included all .Name variables)

[18f9782d79d14caa0534a83a07c4b5134075953edec726bd9d92d84882cba590-json.txt](https://github.com/PLCHome/node-red-contrib-ads/files/5140973/18f9782d79d14caa0534a83a07c4b5134075953edec726bd9d92d84882cba590-json.txt)

PLCHome, 2020-08-28:
Thanks, now I see the error, but can't isolate the problem yet.
I'll take a look at it on the weekend.

hbTecAG, 2020-08-28:
Thank you

No problem, i'm in the mountains till sunday evening, you have all the time ;-)

Thanks for your great Support!

PLCHome, 2020-08-31:
I found a strange recursive call there. I hope that was the problem.
Please check version 1.1.25.

```
removeClient calls : internalRestart
and 
internalRestart calls removeClient 
and 
removeClient calls : internalRestart
```

hbTecAG, 2020-09-02:
Hi Chris,

I tested this now (1.1.25, remove docker container and Start again), same Problem :-(

2 Sep 07:39:06 - [info] Flows gestartet
2 Sep 07:39:07 - [error] [ADS Notification:f8780eaf.a106d] Ads Register Notification '.GHReg1YOut' timeout
2 Sep 07:39:07 - [error] [ADS Notification:565702f7.8b0e5c] Ads Register Notification '.GHZuMe1TIst' timeout
2 Sep 07:39:07 - [error] [ADS Notification:b2797309.dd6ef] Ads Register Notification '.GHReg2Soll' timeout
2 Sep 07:39:07 - [error] [ADS Notification:69029f0a.4504d] Ads Register Notification '.GHReg2TIst' timeout
2 Sep 07:39:07 - [error] [ADS Notification:e851c483.fac338] Ads Register Notification '.GHReg2YOut' timeout
2 Sep 07:39:07 - [error] [ADS Notification:ebbc6ea3.3faf] Ads Register Notification '.GHZuMe2TIst' timeout
2 Sep 07:39:07 - [error] [ADS Notification:676e4157.5031c] Ads Register Notification '.GK_Div17BefehlUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:a0aa17.b757c5e8] Ads Register Notification '.GK_Glocken[1].Status' timeout
2 Sep 07:39:07 - [error] [ADS Notification:31ed69ac.3f9676] Ads Register Notification '.GK_Glocken[2].Status' timeout
2 Sep 07:39:07 - [error] [ADS Notification:8be930c9.9dd7a] Ads Register Notification '.GK_Glocken[3].Status' timeout
2 Sep 07:39:07 - [error] [ADS Notification:635f66d0.c5e8f8] Ads Register Notification '.GK_Glocken[4].Status' timeout
2 Sep 07:39:07 - [error] [ADS Notification:6ef7dbb8.cd18b4] Ads Register Notification '.GK_Glocken[5].Status' timeout
2 Sep 07:39:07 - [error] [ADS Notification:18e939d5.604db6] Ads Register Notification '.GK_Div16BefehlUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:bf54bedc.ef921] Ads Register Notification '.GK_Div01BefehlUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:8a7ceb13.b409a8] Ads Register Notification '.GK_Div02BefehlUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:6ae0b889.181a98] Ads Register Notification '.GK_Div03BefehlUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:ddd73916.d30058] Ads Register Notification '.GK_Div04BefehlUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:14555748.c45da9] Ads Register Notification '.GK_GlockenDauer' timeout
2 Sep 07:39:07 - [error] [ADS Notification:4a10b253.fd525c] Ads Register Notification '.GMdsp1_Preset1SelectUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:d0b4c409.a4f3b8] Ads Register Notification '.GMdsp1_Preset2SelectUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:13402638.1aae9a] Ads Register Notification '.GMdsp1_Preset3SelectUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:bbb8314e.4d911] Ads Register Notification '.GMdsp1_Preset4SelectUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:8be7dd5d.ffd67] Ads Register Notification '.GMdsp1_Mic1plus3dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:f70fd1be.0b47a] Ads Register Notification '.GMdsp1_Mic1minus3dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:18ccefb4.131c9] Ads Register Notification '.GMdsp1_Mic1minus6dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:575c2fcc.3ca48] Ads Register Notification '.GMdsp1_Mic2plus3dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:a81b65ee.ba0cd8] Ads Register Notification '.GMdsp1_Mic2plus0dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:69140439.bdf41c] Ads Register Notification '.GMdsp1_Mic3plus0dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:c60659f2.606d98] Ads Register Notification '.GMdsp1_Mic3minus3dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:a1834e60.c399e] Ads Register Notification '.GMdsp1_Mic3minus6dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:43e85286.734ebc] Ads Register Notification '.GMdsp1_Musik1EnableUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:62428ef0.f7656] Ads Register Notification '.GMdsp1_Musik1SliderWertUi' timeout
2 Sep 07:39:07 - [error] [ADS Notification:7df2da38.587614] Ads Register Notification '.GMdsp1_Musik2EnableUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:bb269882.f88528] Ads Register Notification '.GMdsp1_Musik2SliderWertUi' timeout
2 Sep 07:39:07 - [error] [ADS Notification:3698393a.4e24a6] Ads Register Notification '.oMBeamerLED' timeout
2 Sep 07:39:07 - [error] [ADS Notification:dd8bc572.99d2c8] Ads Register Notification '.GMBe1_Quelle1Uo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:db3966e5.ab1188] Ads Register Notification '.GMBe1_Quelle2Uo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:143e5479.dc6d7c] Ads Register Notification '.GMBe1_MuteUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:38d1697f.49a776] Ads Register Notification '.GMBe1_LampeLebenszeitUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:9781bd55.ebdfc] Ads Register Notification '.gHRaum.raum[0]._calc.Tam' timeout
2 Sep 07:39:07 - [error] [ADS Notification:12f1c83.fce8a38] Ads Register Notification '.gHRaum.raum[0]._calc.PAuf' timeout
2 Sep 07:39:07 - [error] [ADS Notification:b6eb59ae.40ba98] Ads Register Notification '.gHRaum.raum[0]._calc.TAufTotal' timeout
2 Sep 07:39:07 - [error] [ADS Notification:c108a705.4a7fc8] Ads Register Notification '.gHRaum.raum[0]._calc.switch.TdtBeBe[1]' timeout
2 Sep 07:39:07 - [error] [ADS Notification:4cfb192e.b48298] Ads Register Notification '.gHRaum.raum[0]._calc.TAufLauf' timeout
2 Sep 07:39:07 - [error] [ADS Notification:c6a66ea2.6f71] Ads Register Notification '.GHReg0Soll' timeout
2 Sep 07:39:07 - [error] [ADS Notification:dcf64220.10ee4] Ads Register Notification '.GHReg0TIst' timeout
[... cut ...]
2 Sep 07:39:07 - [error] [ADS Notification:ac624b61.f72eb8] Ads Register Notification '.GMdsp1_Mic3plus6dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Notification:91dad737.eead68] Ads Register Notification '.GMdsp1_Mic3plus3dBUo' timeout
2 Sep 07:39:07 - [error] [ADS Output:1e5915a0.2d9b5a] Ads write '.gMemHeizKalender[0][1].valid' timeout
2 Sep 07:39:07 - [error] [ADS Output:ed95f2ac.2e548] Ads write '.oTuerOeff' timeout
2 Sep 07:39:08 - [error] [ADS Output:6eb7a641.453bb8] Ads write '.oLAuss' timeout

PLCHome, 2020-09-02:
I can see the problem now. If I do not deploy individual components but rather the entire flow, the problem arises. But I don't understand the reason.

PLCHome, 2020-09-02:
It's not the Docker, that's fact, sorry :-)

I will rebuild the API and this contrib on promise. I hope that it will be clearer then. the problem must have something to do with the disconnection. I think it's a callback that hasn't been waited for.

Additional question: What do you think if I combine in, out and notification in one node?
Currently I only use Nodered to transfer data from the ioBroker to the PLC via MQTT and I always have to create 2 nodes.

hbTecAG, 2020-09-02:
hmm.. i Only use the notification and out node, not in node (to be honest, I don't quite understand the sense of in)

i think, it's ok, but please, it must be a new node, otherwise i have (and i thank others too) problem's when i make an update on an older version..

hbTecAG, 2020-09-11:
> I will rebuild the API and this contrib on promise. I hope that it will be clearer then. the problem must have something to do with the disconnection. I think it's a callback that hasn't been waited for.

Are you already there? Have you found any bugs yet?

hbTecAG, 2020-10-26:
Did you have found something?

PLCHome, 2020-10-26:
The component is not working properly.
Since I don't feel like node-red and have no time, the project will be closed.

PLCHome, 2020-11-19:
connection improved

closed 2020-11-19 by PLCHome