PLCHome/node-red-contrib-ads #26: support for array and struct variable types

scelle, 2020-03-11:
request for supporting reading and writing of arrays and structure variables

PLCHome, 2020-03-11:
Basically, the component can do that. 
The problem is to create a gui in which to set that. 
I currently don't have the time for that, I'm also not that fit in the node red web gui.

You can currently solve this using a RAW read and a function.

scelle, 2020-03-11:
ok, that's what I'm working on now.  I combine the information from the ads symbols node with the raw data.  
I'll see if I can make time to do some contribution.

PLCHome, 2020-03-11:
I had the idea to create a possibility that you can choose from symbols and types, since these are read from the PLC and are also available in nodered.

maybe you know a node red component where I can learn something like a gui could be constructed

scelle, 2020-03-12:
Combination of using symbols and types as the input for other ads nodes seems like a good idea, 
but combining it all into an easy gui is another thing.
my proposal for an intermediate solution: 

1. extend the symbols ads node:

- add filter field to select one specific symbol name

- add option to "add symbol type info"  

as a result the node could export something that is suitable to use in ADS in, -out and -notification node.

2. extend the ads in, -out, -notification nodes 

As far as I read from the node-ads, this is the info needed to read a struct:
```
 {
    **symname**: '.SOMESTRUCTURE',  
    **bytelength**: [ ads.BOOL, 
                  ads.array(ads.INT,0,9),
                  ads.array(ads.string(10),0,9),
                  ads.array(ads.useLocalTimezone(ads.TIME,false),0,9)
                 ],  
    **propname**: ['value_a.bool',
               'value_a.arrayofint',
               'value_a.arrayofstring',
               'value_a.arrayoftime']
}
```
symname = config.varName
bytelength and propname could be added as an extra config paramater (config.varTypeInfo). This last property should  only be read in case config.varType is of type "STRUCT"

config.varName: (string) override the variable name
config.varType: (string) override the variable type
config.varSize: (integer) the length on RAW and STRING type

PLCHome, 2020-03-12:
yes, but I do not find it good to type such texts on the gui. 
You can give that to the API. 
But you need a GUI to configure it.

193/5000
The gui should be a table to which dynamic lines should be added.
if the type is a type you should be able to select the type again

```
order | valuetype |  type    |   array  | name             | stringlength |  LocalTimezone |startindex   | endindex
1     | TYPE      |          |   false  | value_a          |      %       |        %       |    %        |     %
2     | BOOL      |  value_a |   false  | bool             |      %       |        %       |    %        |     %
3     | BOOL      |  value_a |   true   | arrayofint       |      %       |        %       |    0        |     9
4     | STRING    |  value_a |   true   | arrayofstring    |     10       |        %       |    0        |     9
5     | TIME      |  value_a |   true   | arrayoftime      |      %       |       true     |    0        |     9
```
% = disabled

scelle, 2020-04-27:
Hi, 

I took some time to explore the node and how it works.  
As an excercise I forked your repo and  added functionality for reading **arrays**.
implemented it for the ads-in node 

Seems to work at first sight. (tested arrays of bool, word, strings of defined length)

@PLCHome  what is your working procedure for contributors?  I'm not that familiar with github/contributing...

[image]

PLCHome, 2020-05-05:
can you duplicate it to the ads-notification and ads-write please

scelle, 2020-05-13:
For ads-notification no problem, but  ads-write is a more complex case I think.
I modified the read (and notification) so that you can read the complete array only, not 1 element from an array.  So as a result you get a msg.payload with an array of values.  So it's up to the user to select which value(s) they want to use.  This is acceptable, but an extension should be that you can also define a specific index in the input fields and get 1 value only as a result.

If I would implement ads-write in the same way then you should inject an array of values and so write the complete array.  In most cases, this is not very usefull I think.  It should be possible to write only 1 element from the array.

closed 2020-08-16 by PLCHome