PLCHome/node-red-contrib-ads #27: ADS node stop working after few hours

PeterDezsi, 2020-03-13:
I read some value from twincat 2 system with Siemens iot2040 and forward to these signals to a Siemens 1500 PLC.
The main issue the ADS node is not responsing, after some hours. 
Here is some error messages: 
"Error ADS: Error: This socket has been ended by the other party"
Error stopping node: Close timed out

2020. 03. 13. 8:23:08node: ADSCommStatus
VED_Beckhoff_ADS : msg.payload : Object
object
connectState: 1
connectStateText: "CONNECTIG"
adsState: 0
adsStateText: "INVALID"
**Message from TwinCat2 System:** 
Hint	TwinCAT System (10000)	3/13/2020 8:23:03 AM 843 ms	TCP/IP Connection: Socket to Peer Name: 172.20.77.188, Port: 6873 is connected, current socket will be closed!

But in this case I get the datas from TwinCat. 

Thank you for your help
Best Regards
Peter

PeterDezsi, 2020-03-13:
[flows (20).zip](https://github.com/PLCHome/node-red-contrib-ads/files/4328375/flows.20.zip)

PeterDezsi, 2020-03-13:
[nodes.zip](https://github.com/PLCHome/node-red-contrib-ads/files/4328403/nodes.zip)

PLCHome, 2020-03-13:
It doesn't seem like a big flow.

> Error ADS: Error: This socket has been ended by the other party
Means the PLC closed the soced.

But, after the connection to the PLC has broken down, the component tries to reestablish the connection. Depending on the scenario, this doesn't work very well.

i don't know the performance of the siemens iot. I have over 170 notifications in Node Red on an RP4 4 G with Docker related to a cx1020 PLC with Twincat 2. 19 more containers are running on the Raspberry. The container has been running without problems for more than 2 years. In the beginning everything ran on two PI3b.
[image]

Have you ever tried to increase the ads timeout?
500 is very low value. better 1000 or 1500..

Or the problem is that the timer triggers too many simultaneous requests. 
**Please reduce the number of inquiries at the same time.** 
It can also help not to send constant requests to the PLC. 
The PLC can send the data cyclically via notification, which saves communication.

PLCHome, 2020-03-13:
[image]

You want to be kidding me. 
500ms?
That's too fast for ADS. 

The PLC must end the cycle and transfer the data. No multiread would help either.

Please use the notification component. 
This notifies transmit changes within milliseconds.

closed 2020-03-13 by PLCHome

PeterDezsi, 2020-03-17:
Thank you for your quick answer. 
Before the ADS Input node, I used to the ADS notification and I have same problem. 
Now I changed the cycle time at the ADS node, and also at the timestamp.
So I see today what will be change