PLCHome/node-red-contrib-ads #28: connecting error

HEALproject, 2020-03-18:
Dear PLCHome,

I'm having some problems when using node red to connect to my PLC with contib ads. I hope you can help me.

The goal of the system is to read a %MX memory bit with the raspberry. I am using a BEckhoff PLC and twincat 2. When trying to connect node red to the plc with the "ADS sytem" node, I encounter the following error:
_"Error on connect: check target NetId or routing"_

### This is my setup:
[image]

_(version info below)_

### These are the steps I have made so far:
1. created a new file in TwinCAT System Manager --> choose target --> selected my PLC --> changed netid to 192.168.3.37.1.1
2. created a simple program in TwinCAT PLC control, with two %MX global variables and downloaded it to the PLC
[image]

3. added the I/O devices. I have no Input and Output cards, just the processor. --> see image
[image]

4. appended the PLC project to the PLC configuration

5. under I/O devices I set the route and ADS settings. Since there are 3 devices and I didn't know which one is the right one, I turned them all on.
[image]
[image]

6. on the raspberry I created a node-red flow
[image]

settings ADS connection
[image]

So somehow I am not able to connect to the PLC. I followed your instructions and I also used the Beckhoff manual, but I can't get it done.
 hope you can help me and if you need more information or if it is not clear, please let me know.

Thank you very much in advance!

Kind regards,
Stijn

### Version info
node-red-contrib-ads version: 1.1.22
nodered version: v1.0.3
on which operating system is the Node-RED running: Raspian 10 (buster)
on which hardware system is the Nodered running: Raspberry
PLC type: Beckhoff PLC CX9001-0001
Twincat version of the PLC: v2.10 (build 1342)
Twincat version of the development environment: v2.11 (build 2305)

PLCHome, 2020-03-18:
[image]

sorry, this is the wrong place to enter the Netid. At this point you can link the process images.

Please try to undo this. 
And try to create the link here:
[image]
[image]
[image]
[image]

PLCHome, 2020-03-18:
Please close the ticket if it works.
Or feel free to ask further.

HEALproject, 2020-03-18:
Thank you very much for your quick response, now it works. The ADS System node is working.

Now I am trying to read data from the PLC using the ADS in node, but I do not know how to call the variables. I added some globals variabels which I am trying to read.
[image]

PLCHome, 2020-03-18:
TwinCat refuses to write or read on some variables via ads. 
Global variables must start with a "."  
.led1
.led2
.led3

Maybe you start with a new variable. 
.Test 

Experiments with an out and a notification. 
However, a program must run on the PLC. 
You can also easily access a program variable.

Program.Test

I can also recommend the UI from readme.md

Tiltix7, 2020-03-20:
Hey PLCHome,

I'm also having some trouble connecting to node red. 

My configuration is as follows:
PLC: I'm using twincat 3 (version 3.1 4024.4) and a beckhoff CX5120.
Node-red: latest version running in docker for windows on a PC. Ip-address = 192.168.10.20

I'm unable to create a static route from the PLC to node-red. 

this is my configuration in Node-red
[image]

When I want to create a static route I need to enter a username and password. What do I enter here? 
[image]

Do I need to install twincat on the PC where node-red is running in a docker so I can make a route or am I missing something?

thank you very much in advance.

PLCHome, 2020-03-20:
It is the Windows administration user and his password on the CX.
When shipped from the factory, if you have not changed it, it is:

Default User : Administrator
Default Passwort : 1

Any other administrator is also possible, the rights are important, main users do not work.

closed 2020-03-22 by PLCHome

Tiltix7, 2020-03-24:
Thank you for the quick response. 

I updated the runtime on my PLC and the error dissapeared. Was running twincat 3 (version 3.1 4024.4).