PLCHome/node-red-contrib-ads #30: Problem with changing the variable name - who can help?

goodmike70, 2020-05-28:
Hello,

I have a problem to change the variable name via message command.
I already start a discussion on node red forum but still now the issue is not solved.
[https://discourse.nodered.org/t/ads-in-change-var-name/27210/8](url)

Maybe here someone can help - Thanks in advance!

PLCHome, 2020-05-31:
You have to set msg.config.varName. Config is at the same level as payload in the message. 
You can actually influence everything about the message
```
{"payload":false,
 "config":{"varName":".gRMEgWohnzimmer.Alarmstatus",
           "varType":"BOOL",
           "timezone":"UNCHANGED"}}
```

closed 2020-05-31 by PLCHome

goodmike70, 2020-06-01:
@PLCHome Sorry, wehre I have to enter this? In the "ADS in" node itself? If yes at which position in the config field? Or did I have to put a "change" node in front and link the output of these to the input of the "ADS in"? If this is true, what I have to configure in the "change" node?
Sorry again for this, maybe stupid question, but I'm a total newbie...

PLCHome, 2020-06-02:
I took a look at the thread.

First of all: ADS-IN with an timed Inject is never advisable. Always use ADS-Notification. This relieves the PLC

The Inject only can set msg.payload.
To set the config use an function or the change node:

```
[{"id":"ca8765a.91f9d98","type":"inject","z":"e1aabefe.98186","name":"","topic":"","payload":"123","payloadType":"num","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":390,"y":140,"wires":[["eebaca22.926af8"]]},{"id":"eebaca22.926af8","type":"change","z":"e1aabefe.98186","name":"","rules":[{"t":"set","p":"config","pt":"msg","to":"{\"varName\":\".RolloFunktionStatus_NodeRed[3]\"}","tot":"json"}],"action":"","property":"","from":"","to":"","reg":false,"x":610,"y":140,"wires":[["28ec80c.a7ce28"]]},{"id":"28ec80c.a7ce28","type":"debug","z":"e1aabefe.98186","name":"","active":true,"tosidebar":true,"console":false,"tostatus":true,"complete":"true","targetType":"full","x":810,"y":140,"wires":[]}]
```

Here is my productive knx/eib to twincat converter. The knx/eib data of the smoke detectors come in as binary data and are then converted. The variable is determined based on the knx/eib address.
```
[{"id":"2f863991.ca4b26","type":"knx-in","z":"85c9eed2.b2411","name":"","controller":"fb693d65.d6081","x":150,"y":140,"wires":[["b8206fc9.f2596","931fdb43.f54348"]]},{"id":"b8206fc9.f2596","type":"function","z":"85c9eed2.b2411","name":"KNX to PLC","func":"const room = {\n    '1/1':'.gRMKgFlur',\n    '1/2':'.gRMKgHWR',\n    '1/3':'.gRMKgAbstellraum',\n    '1/5':'.gRMKgWohnraum',\n    '2/1':'.gRMEgFlur',\n    '2/5':'.gRMEgWohnzimmer',\n    '2/6':'.gRMEgHWR',\n    '3/3':'.gRMOgSchlafzimmer',\n    '3/4':'.gRMOgZimmerLinks',\n    '3/5':'.gRMOgZimmerRechts',\n    '3/1':'.gRMOgFlur',\n    '4/2':'.gRMDsZimmer',\n}\nconst type = {\n    '/4':{\n        varname: '.Alarmstatus',\n        vartyp: 'BOOL',\n        func: (buf)=>{return !(!buf.readUInt8(0))},\n    },\n    '/9':{\n        varname: '.Verschmutzungsgrad',\n        vartyp: 'LREAL',\n        func: (buf)=>{return Math.round((buf.readUInt8(0)*1000.0)/255.0)/10.0},\n    },\n    '/10':{\n        varname: '.Batteriespannung',\n        vartyp: 'LREAL',\n        func: (buf)=>{return RealfromBuffer(buf)/1000}, \n    },\n    '/11':{\n        varname: '.Temperatur',\n        vartyp: 'LREAL',\n        func: (buf)=>{return RealfromBuffer(buf)}, \n        reset: {\n            payload: new Date(0),\n            config: {\n                varName: '.MAXNEXTUPDATE',\n                varType: 'TIME',\n                timezone: 'UNCHANGED',\n            },\n        },\n    },\n    '/14':{\n        varname: '.Fehlfunktion',\n        vartyp: 'BOOL',\n        func: (buf)=>{return !(!buf.readUInt8(0))}, \n    },\n}\n\nfunction ldexp(mantissa, exponent) {\n    return exponent > 1023 ? mantissa * Math.pow(2, 1023) * Math.pow(2, exponent - 1023): exponent < -1074 ? mantissa * Math.pow(2, -1074) * Math.pow(2, exponent + 1074): mantissa * Math.pow(2, exponent);\n}\nfunction RealfromBuffer(buf) {\n    var sign     =  buf[0] >> 7;\n    var exponent = (buf[0] & 0b01111000) >> 3;\n    var mantissa = 256 * (buf[0] & 0b00000111) + buf[1];\n    mantissa = (sign == 1) ? ~(mantissa^2047) : mantissa;\n    return ldexp((0.01*mantissa), exponent);\n}\n\nif (msg.payload.dstgad) {\n    var adress = msg.payload.dstgad.split('/')\n    var varroom = room[''+adress[0]+'/'+adress[1]]\n    var vartype = type['/'+adress[2]]\n\n    if (varroom && vartype){\n        //msg.var = varroom+vartype\n        //msg.val = vartype.func(msg.payload.value)\n        if (vartype.reset) {\n            var resetmsg = Object.assign({},vartype.reset)\n            resetmsg.config.varName = varroom+resetmsg.config.varName\n            node.send(resetmsg)\n        }\n        var outmsg = {\n            payload: vartype.func(msg.payload.value),\n            config: {varName: varroom+vartype.varname,\n                varType: vartype.vartyp,\n                timezone: 'UNCHANGED' }\n            }\n        }\n        \n        return outmsg\n}\nreturn null\n","outputs":1,"noerr":0,"x":330,"y":140,"wires":[["eed4ac15.8369b","bde0b5ee.21e988"]]},{"id":"eed4ac15.8369b","type":"debug","z":"85c9eed2.b2411","name":"","active":true,"tosidebar":true,"console":false,"tostatus":true,"complete":"true","x":530,"y":100,"wires":[]},{"id":"bde0b5ee.21e988","type":"ADS Output","z":"85c9eed2.b2411","name":"","datasource":"1aefd369.53154d","varName":".gRMEgHWR.Temperatur","varTyp":"LREAL","outValue":"payload","varSize":"","timezone":"UNCHANGED","topic":"","x":590,"y":180,"wires":[]},{"id":"fb693d65.d6081","type":"knx-controller","z":"","ipAddr":"","ipPort":"","physAddr":""},{"id":"1aefd369.53154d","type":"ads-connection","z":"","host":"","amsNetIdTarget":"","amsNetIdSource":"","port":"","amsPortSource":"","amsPortTarget":"","adsTimeout":"1000"}]
```

goodmike70, 2020-06-07:
Thanks for your support and example. Now it works fine!