PLCHome/node-red-contrib-ads #33: ADS Read to indexOffset and indexGroup

florian-adelt, 2020-08-14:
Hello,
thanks for your great node-red Package.
I'm using it on a customer project to fetch variables from the PLC and push it to the cloud.

Now we have to read Values directly via ADS from a connected Sensor. The Sensor is connected to the PLC via IO-Link.

Problem is, that we don't want to read the values in a PLC variable/symbol. We wanna access directly and we have only the address in the datasheet of the sensor.

I found a solution with the ads-api:
```
amsNetIdTarget: "5.54.245.250.4.5", //last two blocks is IO-Link Master instead of PLC
amsPortTarget: 4097 //Port increments for each sensor on the master.
```

The following code fetches the Values of the Address:
```
var myHandle = {
    indexOffset: 0x00120000,
    indexGroup: 0xF302, 
    bytelength: 18,
    propname: 'value'
}

var client = ads.connect(options, function () {
    this.read(myHandle, function (err, result) {
        if (err) console.log(err)
        console.log(result)
    })
    console.log("Connected")
})
```

The result is an Array with the Value. In this case the Sensor returns the String of the Manufacturer "SICK AG".

I could create my own node-red package to make a read node with the configurable values indexOffset, indexGroup, bytelength. But i think it would be better placed into your code.
Are you interested to implement this feature in the next months? If not, would you add my code if it is good?

If not, i will create my own package for that.

Looking forwad to hear from you.

Florian (Germany)

PLCHome, 2020-08-16:
Feel free to make a Pull request

PLCHome, 2020-11-19:
Please test version 1.1.28 and let me know if it works

closed 2020-11-19 by PLCHome