PLCHome/node-red-contrib-ads #34: Error ADS: Error: connect ETIMEDOUT

tpatora, 2020-08-31:
Hi,
thanks for ads package.

My config:
Node-RED version: v1.1.3
Node.js  version: v14.8.0 [Windows_NT 10.0.18363 x64 LE]

Problem:
When the ads network is inaccessible and there is no connection to the other host (TwinCAT 3), an error occurs after starting node-red:

 _[error] [ads-connection:29289c71.cd9d64] Error ADS: Error: connect ETIMEDOUT 10.1.12.84:48898
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1144:16) {
  errno: -4039,
  code: 'ETIMEDOUT',
  syscall: 'connect',
  address: '10.1.12.84',
  port: 48898
}

Unfortunately, the system does not respond, all flows stop.
After restarting the node-red timeout occurs after several seconds and stops node-red again.

Logging level trace didn`t give me a hint. Node-Red just stoped after error message.

Any idea?

PLCHome, 2020-09-01:
Ich verstehe den issue nicht. 
Wenn die SPS nicht läuft oder kein Netzwerk vorhanden ist, kann sich deise Node nicht verbinden, oder was willst du mir sagen?

PLCHome, 2020-09-01:
I don't understand the issue.
If the PLC is not running or if there is no network, the node cannot connect, or what do you want to tell me?

tpatora, 2020-09-01:
I'm sorry that I didn't make myself clear enough.

In the case of the most current version, the whole node-red (flow, ui) was
suspended in case of connection problem.
It looks like in case of wrong connection, node-red hangs completely and
only restarting the whole node-red helped.
This seems to be a problem for node-red itself, especially for network
connections.

In the LTS version, if there is no connection, it tries to reconnect after
a few seconds and is ok.

I'd like to take this opportunity to ask about the possibility of saving
the whole array and not individual items.
Reading the arrays works perfectly, even multidimensional arrays, but I did
not find the writing.

Great job with ads! You are great!

PLCHome, 2020-10-26:
The component is not working properly.
Since I don't feel like node-red and have no time, the project will be closed.

PLCHome, 2020-11-19:
connection improved

closed 2020-11-19 by PLCHome