PLCHome/node-red-contrib-ads #36: ADS System - DISCONNECTING state lock

tpatora, 2020-09-06:
Hi,
Thanks for the super module ads

I'm trying to find the cause of the problem but I can't manage without your help.

I try different Node-Red (PC<->TwinCAT PC) and (RaspberryPi <->TwinCAT PC) and always get the same result.

When I restart TwinCAT (computer, reload program), ADS-System change state to **DISCONNECTING** and hung. Only restart whole node-red change to RUN state. 

When system is **RUN** state I get:
_connectState: 2
connectStateText: "CONNECTED"
adsState: 5
adsStateText: "RUN"
majorVersion: 3
minorVersion: 1
versionBuild: 1917
version: "3.1.1917"
deviceName: "Plc30 App"
symTab: 40_

**When restart ADS**

 _[error] [ads-connection:53de57ab.a40fb8] Error ADS: Error: This socket has been ended by the other party
    at Socket.writeAfterFIN [as write] (net.js:455:14)
    at Object.runCommand (/home/pi/.node-red/node_modules/node-ads-api/lib/ads.js:1025:17)
    at Object.writeCommand (/home/pi/.node-red/node_modules/node-ads-api/lib/ads.js:890:14)
    at Object.releaseSymHandle (/home/pi/.node-red/node_modules/node-ads-api/lib/ads.js:828:18)
    at Object.releaseSymHandles (/home/pi/.node-red/node_modules/node-ads-api/lib/ads.js:813:22)
    at Object.end (/home/pi/.node-red/node_modules/node-ads-api/lib/ads.js:180:21)
    at EventEmitter.ads.adsClient.end (/home/pi/.node-red/node_modules/node-ads-api/lib/ads.js:56:16)
    at internalRestart (/home/pi/.node-red/node_modules/node-red-contrib-ads/ads-connection.js:385:24)
    at Timeout.<anonymous> (/home/pi/.node-red/node_modules/node-red-contrib-ads/ads-connection.js:147:11)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7) {
  code: 'EPIPE'
}_

and then I get:

_connectState: 3
connectStateText: "DISCONNECTING"
adsState: 0
adsStateText: "INVALID"
majorVersion: 3
minorVersion: 1
versionBuild: 1917
version: "3.1.1917"
deviceName: "Plc30 App"
symTab: 40_

I wait several minutes to change state to RUN and nothing happened.
Only restart fix this issue.

MrMoo1337, 2020-09-08:
That is the same issue i have, unfortunately the issue was closed with an help.

Maybe my temporary solution will help you - 
if the ads system state is in disconnecting a trigger causes a delay of a predefined time. If the disconnecting state exist longer than this time a .cmd fall will be executed over the "exec" node.

this cmd file includes following:

"start /min taskkill /IM cmd.exe /FI "WINDOWTITLE eq node-red*" && start /min C:\Users\murat\AppData\Roaming\npm\node-red.cmd
EXIT"

This cmd file opens a cmd prompt in the background,  eliminates the existing node-red prompt and restarts it. 

I hope this helps you a little bit

MrMoo1337, 2020-09-08:
That is the same issue i have, unfortunately the issue was closed without a solution.

Maybe my temporary solution will help you -
if the ads system state is in disconnecting a trigger causes a delay of a predefined time. If the disconnecting state exist longer than this time a .cmd fall will be executed over the "exec" node.

this cmd file includes following:

"start /min taskkill /IM cmd.exe /FI "WINDOWTITLE eq node-red*" && start /min C:\Users\XXX\AppData\Roaming\npm\node-red.cmd
EXIT"

This cmd file opens a cmd prompt in the background, eliminates the existing node-red prompt and restarts it.

I hope this helps you a little bit

massimo-pizzardi, 2020-09-10:
Hello, I have the same problem. Restarting PLC causes disconnection from NodeRed (Raspberry). Can you help me to force a restart of NodeRed? thanks in advance
Massimo

PLCHome, 2020-10-26:
The component is not working properly.
Since I don't feel like node-red and have no time, the project will be closed.

PLCHome, 2020-11-19:
connection improved

closed 2020-11-19 by PLCHome