PLCHome/node-red-contrib-ads #38: RangeError: Maximum call stack size exceeded

hbTecAG, 2020-10-26:
Hi

i Have another Problem: 

26 Oct 09:15:23 - [red] Uncaught Exception:
26 Oct 09:15:23 - RangeError: Maximum call stack size exceeded
    at Object.releaseSymHandles (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:1)
    at Object.<anonymous> (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:814:25)
    at Object.releaseSymHandle (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:831:44)
    at Object.releaseSymHandles (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:813:22)
    at Object.<anonymous> (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:814:25)
    at Object.releaseSymHandle (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:831:44)
    at Object.releaseSymHandles (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:813:22)
    at Object.<anonymous> (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:814:25)
    at Object.releaseSymHandle (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:831:44)
    at Object.releaseSymHandles (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:813:22)
mbus_sendrecv_request: No M-Bus response frame received.
26 Oct 09:15:55 - [info]

Welcome to Node-RED
===================

26 Oct 09:15:55 - [info] Node-RED version: v1.1.3
26 Oct 09:15:55 - [info] Node.js  version: v12.16.1

V1.1.25 von ads nodes..

PLCHome, 2020-10-26:
The component is not working properly.
Since I don't feel like node-red and have no time, the project will be closed.

PLCHome, 2020-11-19:
connection improved

closed 2020-11-19 by PLCHome