PLCHome/node-red-contrib-ads #39: Project adoption

gfcittolin, 2020-10-27:
Hi there,

I've just discovered this project last week, and while revisiting the repository today, I noticed the deprecation message, and I feel very sorry for that. I know it can be very time-intensive to maintain such a project. Even though, I've tested the node and have indeed had some problems making it work :wink: 

I'm the main maintainer of the `node-red-contrib-s7` (for communicating with Siemens PLCs) and `node-red-contrib-cip-eth-ip` (for Rockwell ones), among others, and I wouldn't like to see this dying. I'd like to help maintaining the project, or even taking the ownership if you don't have any interest at all in maintaining it. Of course I could just fork it, but I think it works better just continuing your work. 

Would you be willing to move in such direction?
Despite everything, thank you for the contribution to the Node-RED community!

scelle, 2020-10-28:
Hi, indeed, sad to read the message.  Always been very satisfied about this node.  Indeed the first steps to setup connection can be hard sometimes.  I don't know enough about the system but I suppose the underlying ads node might be the cause of some of the latest problems.   
But anyway,  once running, this node works flawless and without issues as for my cases.    So I'm convinced it's worth maintaining the project!

PLCHome, 2020-10-28:
For some people the most important thing seems to be deploying. And that doesn't work properly.

gfcittolin, 2020-10-29:
Sure there may be some improvements, but my concern right now is about the project's maintainability and status. I'm willing to invest quite some time on the improvement of both the node and the underlying library, and therefore I'd like to know whether I'll be able to keep the project alive, or if all this is pointless and you've decided take the project down anyway.

For now, I've checked the code base and there may be some improvements to be done there. I'm going to fork this on the next days after I finish some other projects I'm working on, and until then we have some time to decide how we'll manage this project

brechtgijsens, 2020-10-29:
I'm also willing to invest some time into the project, did my thesis with it and it worked like a charm. But before I do this I would like to know if its as handy to develop this without a real PLC and just use the software plc from my pc. I know this works but maybe its not as good for the development of the nodes.

closed 2020-10-29 by PLCHome