PLCHome/node-red-contrib-ads #40: Trying to connect two Beckhoff PLCs from one Node-Red... Failing

BArweiler, 2020-12-03:
Each ads-connection configuration for itself is running okay (1 ads connection, the other deleted)... 
Running both leads on both system-nodes to cyclic error and re-establishing of the connections.
I have different source ports in the configurations.
Node-Red is running in a docker container with the newest Node-Red and the newest version of your nodes.
I downgraded your nodes to 1.1.22 and it seems to be running ....

PLCHome, 2020-12-03:
Client field in data-source must be empty when docker is not in host or vnet mode.

BArweiler, 2020-12-03:
I tried empty and IP in client id and I put the container even in net= host
That helped at this point not.
But I try all combination with new node version again

BArweiler, 2020-12-07:
Hello, I tried the newest version again, and it fails with 2 PLC connections ... Which is running with 1.1.22 and 1.1.24.
Same setup with docker (net = host) and in 1.1.24 even in clientid the IP address... 
In 1.1.22 the topic from the Input port was not forward to the outport.. This is okay in 1.1.24... So something in the newest changed with the handling of multiple PLCs......

For the moment 1.1.24 is helping me.. 
Thx for the nice project

closed 2020-12-07 by PLCHome