PLCHome/node-red-contrib-ads #41: CX9020 and TwinCAT 2

xavier-1197, 2020-12-09:
Hello PLCHome,

I have a problem with the node-red-contrib-ads module. I can read a global variable in my CX9020 PLC (with TwinCAT 2) but I can't read a variable in a program. I write the right path

Can you help me please thanks

[image]

xavier-1197, 2020-12-09:
Here you can see my configuration

[image]

PLCHome, 2020-12-09:
Compteur is not a function block under MAIN. Compteur is a program so "MAIN." leave out and it should go "Compteur.wCTUCv". The writing will blocked by twincat 2. Reading goes.

PLCHome, 2020-12-09:
You can see it with ADS symbols. There are flow examples in read.me.

xavier-1197, 2020-12-09:
Thanks a lot @PLCHome 
That works.
The example 
[image]
took me down the wrong path.

It makes perfect sense now

closed 2020-12-09 by xavier-1197