PLCHome/node-red-contrib-ads #42: Ads Read to IndexOffset

florian-adelt, 2020-12-14:
Related ISSUE #33 

Hello @PLCHome,
thank you! Sorry for the delayed answer - i am rarely on the site of my customer to test.

I tested it and get the error "is nan is not defined." 

Also i creadet my own package "node-red-contrib-ads-read", which works fine. I will delete it as soon as your package ist working on our installation.

In picture you can see my flow.
[image]

PLCHome, 2020-12-14:
Unfortunately, I can't see why.
Can you post the flow as json.

closed 2021-01-22 by PLCHome

kapperi, 2021-02-01:
Hi,

I have this same problem. There´s the flow below.

My problem is that I have so old PLC that it doesn´t allow variable names to fetch or send.

`[{"id":"ce4fcb18.11fd18","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"9866c904.2d2558","type":"ADS Output","z":"ce4fcb18.11fd18","name":"out","datasource":"25a3a4fb.8bfa2c","useIndex":true,"varName":"","indexGroup":"0xF030","indexOffset":"0x3F2","varTyp":"BOOL","outValue":"payload","varSize":"","timezone":"UNCHANGED","topic":"","x":690,"y":500,"wires":[]},{"id":"36518ec9.019692","type":"inject","z":"ce4fcb18.11fd18","name":"","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"0","payloadType":"num","x":420,"y":500,"wires":[["93ae1ae3.2c8428","9866c904.2d2558","b65769cf.eca3f8"]]},{"id":"93ae1ae3.2c8428","type":"debug","z":"ce4fcb18.11fd18","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":720,"y":420,"wires":[]},{"id":"b65769cf.eca3f8","type":"ADS In","z":"ce4fcb18.11fd18","name":"in","datasource":"25a3a4fb.8bfa2c","useIndex":true,"varName":"","indexGroup":"15","indexOffset":"15","varTyp":"BOOL","inValue":"payload","varSize":"","varLowIndex":"","varHighIndex":"","useInputMsg":false,"isArray":false,"timezone":"UNCHANGED","topic":"","x":700,"y":560,"wires":[["a83483ec.966b"]]},{"id":"a83483ec.966b","type":"debug","z":"ce4fcb18.11fd18","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":910,"y":560,"wires":[]},{"id":"25a3a4fb.8bfa2c","type":"ads-connection","host":"192.168.1.20","clientIP":"192.168.1.19","amsNetIdTarget":"192.168.1.20.1.1","amsNetIdSource":"192.168.1.19.1.1","port":"48898","amsPortSource":"32905","amsPortTarget":"801","adsTimeout":"500"}]`

machadotiago, 2021-07-27:
Did someone ever managed to solve this issue? I have the same  "is nan is not defined." issue when reading using index. When I read using the varname it works fine.

machadotiago, 2021-07-28:
I managed to make it run. I found the bug and I'll submit a PR.