PLCHome/node-red-contrib-ads #46: Problem with 2 ADS Connections

hbTecAG, 2021-01-29:
Hello there

I have a Problem, that if i have 2 ADS connection Nodes active.
i think, since V1.1.29, it is no more working. I updated from 1.1.25. I have become often an Error, that Symbols are not found: 

`Ads write '.GHReg13Soll' Error: symbol not found    at getError (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:1800:13)    at Object.getWriteReadResult (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:1074:13)    at Object.analyseResponse (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:288:28)    at Object.checkResponseStream (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:226:25)    at Socket.ads.dataCallback (/hbvisu-data/node_modules/node-ads-api/lib/ads.js:155:27)    at Socket.emit (events.js:311:20)    at Socket.EventEmitter.emit (domain.js:482:12)    at addChunk (_stream_readable.js:294:12)    at readableAddChunk (_stream_readable.js:275:11)    at Socket.Readable.push (_stream_readable.js:209:10)`

The Symbol is declared as a Global Variable and works fine. 
What do I have to look out for if I have two config nodes?

ADS Version: 1.1.29
Node-RED:  1.1.3
Node.js: 12.16.1
Working in a Dockercontainer

I think, it's a problem with the ports in the configuration node..? What can/must i change there?
Both PLC's are TwinCAT 2.11.X. I need Node-RED to share some Variables and at the same time, visualize them..

First PLC config:
[image]

Second PLC config:
[image]

Do you have any idea / solution for me? 
Thanks alot

PLCHome, 2021-01-29:
I do not know that. Try the source port on 32906 with one of the two PLC connections.

hbTecAG, 2021-02-01:
Hello
Unfortunately the same problem. as soon as I delete the config node it works, as soon as I add a second one (also with port 32906) it no longer works.

scelle, 2021-02-01:
Hi, 
Same issue here.  I also have a NR project with 2 PLC connections.  with v1.1.29 it is not working.    Unfortunately, had no time to look at it in depth, so for now I'm staying on v 1.1.24.
As far as it may be usefull, my config is like this:
1st PLC config
[image]
2nd PLC config
[image]

So my NR is running on PLC 1 and needs to make ads connection to a local plc runtime (TC3) and to another one (TC2) in the network.

Sorry I can't be of any more help for now.

hbTecAG, 2021-04-27:
@vdwpsmt 
Have you been able to analyze the problem in more detail in the meantime?

is there any prospect of a bugfix?

vdwpsmt, 2021-04-27:
@hbTecAG 
Sorry, not from my side.  I switched over to OPC-UA for the specific project I mentioned (for stability reasons).  (Although I still believe the ADS communication is the more lightweight, performant and easy solution).

closed 2023-05-15 by PLCHome