PLCHome/node-red-contrib-ads #47: ADS disconnects roughly every 15s

henne959, 2021-02-28:
I'm running it on a Raspberry Pi 3b+ with just a few reads and a dashboard with buttons to write BOOLs. The Pi is connected via WLAN to the home network where the PLC CX8190 (TwinCat 3.1.4321) is connected via LAN. 

Every ~15s the connectStateText is changing from connected --> disconnected --> connected.
The adsStateText changes from run --> invalid --> run

The disconnected state lasts for about 5s.
I don't get an other error. 
The only cyclic read is with the ADS System node. 

What can cause this?

PLCHome, 2021-02-28:
It looks like the Net-ID and IP are not stored correctly in the PLC. 

It's well explained here.
https://github.com/dkleber89/ioBroker.beckhoff
Since it uses the common API, the process is identical.

henne959, 2021-03-19:
Sorry for the late reply!

I forgot to check the "Symbole erzeugen" property. 

Now it runs smoothly, thanks!

closed 2021-03-19 by henne959