PLCHome/node-red-contrib-ads #53: Using node state for var content and connection state

KingAndi, 2023-07-17:
Hallo!

Is it possible to use the node state to show the connection state (the small rectangle below the node in e.g. green/grey/red) and to show the content of the variable? This would be perfect so it's not necessary to add extra debug nodes...

Best regards, Andreas

PLCHome, 2023-07-17:
ADS System:
```

msg.payload : Object
{ "connectState":2,
  "connectStateText":"CONNECTED",
  "adsState":5,
  "adsStateText":"RUN",
  "majorVersion":2,
  "minorVersion":11,
  "versionBuild":2605,
  "version":"2.11.2605",
  "deviceName":"TCatPlcCtrl",
  "symTab":138
}

```
There is "connectState" and  "connectStateText" as output

closed 2023-07-17 by PLCHome

KingAndi, 2023-07-17:
Hallo!

That was not the intention of my post. I suggest to use the state each node to show the content of it's variable it is holding...
Like the debug node, where you can show below the node the debug info...

PLCHome, 2023-07-17:
Ok make a merge request. For me this Project is closed, take a look in the readme.